facebookresearch/habitat-sim #2175: issues with

deBrian07, 2023-08-04:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

deBrian07, 2023-08-04:
The following error occurred during the execution of the line 'sim = habitat_sim.Simulator(cfg)':
'''
Platform::WindowlessEglApplication::tryCreateContext(): cannot get default EGL display: EGL_BAD_PARAMETER
WindowlessContext: Unable to create windowless context
'''

I'm currently using the headless version since I'm on a cloud GPU cluster (runpod.io)

deBrian07, 2023-08-05:
Can anyone help please?

flxinxout, 2023-10-23:
I have the same error please

aclegg3, 2023-10-23:
@0mdc

0mdc, 2023-10-25:
This is probably due to a driver or configuration issue.

I would suggest verifying that the drivers are properly installed and up-to-date. With NVIDIA, you can use `nvidia-smi` to validate that the driver is installed.

If the issue persists, you can run Habitat with the environment variables `MAGNUM_GPU_VALIDATION=ON` and `MAGNUM_LOG=verbose`. This should reveal additional information about the issue in the logs.

flxinxout, 2023-12-25:
> This is probably due to a driver or configuration issue.
>
> I would suggest verifying that the drivers are properly installed and up-to-date. With NVIDIA, you can use `nvidia-smi` to validate that the [...]

My problem was solved by adding more storage to my VM (so now 70GB)