facebookresearch/habitat-sim #2191: How to create a separate pathfinder instance for inflating the map for planning?

sai-prasanna, 2023-08-22:
I want to create a pathfinder where the agent's radius is inflated. I want to use this for path planning. But I want collisions to be computed with the original navmesh where I set the correct robot radius. Is there a way to do this?

aclegg3, 2023-08-22:
Hey @sai-prasanna, good question.

Yes, you can manage as many PathFinder instances as you like.
I believe the current implementation typically uses `sim.pathfinder` for most internal callbacks and collision checks, so you'll want to keep the default there.
You can create your own instance to use specifically as you like by constructing it and managing the calls yourself.
For example to create an inflated NavMesh for your own use:
```
#construct a new PathFinder instance
pf = habitat_sim.nav.PathFinder()

#copy the current PathFinder NavMeshSettings and inflate the radius
modified_settings = sim.pathfinder.nav_mesh_settings
inflated_navmesh_radius = modified_settings.agent_radius
modified_settings.agent_radius += 0.05

#here I pass my new PathFinder instance, pf, to the recompute function
assert sim.recompute_navmesh(pf, modified_settings), "failed to recompute navmesh."
```

aclegg3, 2023-08-22:
Note that because [habitat_sim.nav.PathFinder.nav_mesh_settings](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#nav_mesh_settings) is a get only property, it is provided as a copy, so you don't need to worry about modifying the original.

aclegg3, 2023-08-22:
If you have any further questions on this topic, please let me know, otherwise you can close the issue. :slightly_smiling_face:

sai-prasanna, 2023-10-19:
Hi, sorry for the late reply. Indeed creating a new instance works!

closed 2023-10-19 by sai-prasanna