facebookresearch/habitat-sim #2196: Pass 'Combine redundant instructions' is not initialized. Verify if there is a pass dependency cycle.

gudehhh666, 2023-08-28:
## Habitat-Sim version
v0.2.2
When I test the habitat using python, this issue will happen

`python examples/viewer.py --scene /media/cv/data/wang/habitat/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

```
[00:35:57:845210]:[Assets] ResourceManager.cpp(855)::loadRenderAsset : Loading general asset named: /media/cv/data/wang/habitat/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
[00:35:57:845221]:[Assets] ResourceManager.cpp(1641)::ConfigureImporterManagerGLExtensions : Importing Basis files as ASTC 4x4.
**Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.**
Required Passes:
段错误 (核心已转储)

```
Please someone can help me out!!!

aclegg3, 2023-09-06:
Hey @gudehhh666 
Strange error. I haven't seen that before.

First check: I see v0.2.2, is this installed from conda? 
What version is cloned?
Most likely issue is something related to cloned vs. installed versions mismatched. Since examples/ is not part of the module, this can happen.

Other issues could be system related. 

In either case, this was not a general issue on that version and is not an issue on `main` currently, so it isn't something we can fix. Happy to help provide suggestions with more information though.

xuboshen, 2023-12-09:
Hi, I've met similar issue, do anyone have some ideas?
my environment is inside a docker on remote server without display: habitat_sim v0.2.5, cuda: 11.7, ubuntu22.04
building from source: 
```
git clone --branch v0.2.5 https://github.com/facebookresearch/habitat-sim.git
python setup.py install --with-cuda --bullet # since I've setup the X11 forward, the view should be shown on my local pc.

```
following the testing process to download the scenes and objects,

while run `python ./examples/viewer.py --scene ../check_habitat_sim/scene_datasets/habitat-test-scenes/skokloster-castle.glb`,

the error occurs:
```
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Platform::GlfwApplication: physical DPI scaling Vector(1, 1)
Renderer: llvmpipe (LLVM 12.0.0, 256 bits) by Mesa/X.org
OpenGL version: 4.5 (Core Profile) Mesa 21.2.6
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_texture_view
    GL_ARB_framebuffer_no_attachments
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-implementation-color-read-format-dsa-explicit-binding
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
GL::Context: enabling GPU validation
PluginManager::Manager: duplicate static plugin StbImageImporter, ignoring
PluginManager::Manager: duplicate static plugin GltfImporter, ignoring
PluginManager::Manager: duplicate static plugin BasisImporter, ignoring
PluginManager::Manager: duplicate static plugin AssimpImporter, ignoring
PluginManager::Manager: duplicate static plugin AnySceneImporter, ignoring
PluginManager::Manager: duplicate static plugin AnyImageImporter, ignoring
[11:09:15:915302]:[Warning]:[Assets] ResourceManager.cpp(353)::loadSemanticSceneDescriptor : SSD File Naming Issue! Neither SceneInstanceAttributes-provided name : `../check_habitat_sim/scene_datasets/habitat-test-scenes/skokloster-castle.scn` nor constructed filename : `../check_habitat_sim/scene_datasets/habitat-test-scenes/info_semantic.json` exist on disk.
Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:
Segmentation fault (core dumped)
```

after check, the error occurs at line 366 in example/viewer.py: `habitat_sim.Simulator(self.cfg)`

appreciate for any help in advance !

HW140701, 2024-10-17:
I also encountered this problem, how to solve it

Yuxin916, 2024-12-05:
Hi how did you solve the PluginManager::Manager: duplicate static plugin problem? Thank you.

wu5bing, 2024-12-27:
> ## Habitat-Sim version
> v0.2.2 When I test the habitat using python, this issue will happen
>
> `python examples/viewer.py --scene [...]

I've met the same same issue in `v0.3.2`. I solved it by downgrading to `v0.3.1`. 

`conda install habitat-sim==0.3.1 withbullet -c conda-forge -c aihabitat`

I dont know why it works, both two version install by conda. But it solves my issue.