facebookresearch/habitat-sim #2197: Holes with fixed grid-like patterns in the rendered RGB observations

jeasinema, 2023-08-29:
## Habitat-Sim version
v0.1.6 (actually, it is a [customized build](https://github.com/Ram81/habitat-sim) by @Ram81 :) )

## ❓ Questions and Help

I am replaying some habitat-web demonstrations of the [MP3D pick and place task] (https://github.com/Ram81/habitat-imitation-baselines#downloading-human-demonstrations-dataset). But sometimes (yes, it won't always happen and I don't quite know what is the occasion, sorry!) the rendered reply video I got has lots of holes with weird grid-like patterns on it, [demo](https://drive.google.com/file/d/1BXBz1jFV4n3rJA5siNrybPVfnUd5KprI/view?usp=sharing) 

Does this seem familiar to you guys? Sorry for not being able to provide more details, but any insights will be much appreciated! 

I use an A100 GPU for rendering. Feel free to let me know if you need more information.

Best,
XM

jeasinema, 2023-08-29:
this might be relevant, looking into it now https://github.com/facebookresearch/habitat-sim/issues/1987

jeasinema, 2023-08-29:
Can confirm bumping to CUDA 11.8 solves this issue! PS: if you work on a GPU cluster without root:

1. Install CUDA using anaconda

```bash
conda install cudatoolkit=11.8 -c anaconda
```

2. Override the system CUDA with `LD_LIBRARY_PATH`

```bash
export LD_LIBRARY_PATH=<path_to_anaconda>/pkgs/<cudatoolkit_path>/lib
```

closed 2023-08-29 by jeasinema

mahnoor-fatima-saad, 2024-01-07:
I'm currently having the same issue
Bumping cuda didnt help 
Any leads?