facebookresearch/habitat-sim #2205: ImageExtractor Error

solomon-ma, 2023-09-18:
## Habitat-Sim version
v0.2.5

## System Info
Ubuntu 20.0.4

## 🐛 Bug

##Code only about ImageExtractor
import torch
from torch.utils.data import DataLoader, Dataset
from torchvision import utils
from torchvision.transforms import ToTensor
import habitat_sim.registry as registry

from habitat_sim.utils.data import ImageExtractor, PoseExtractor

# Replace with the path to your scene file
# SCENE_FILEPATH = 'data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb'
scene_file = 'data/scene_datasets/habitat-test-scenes/apartment_1.glb'

extractor = ImageExtractor(scene_file, 
                           img_size=(512, 512),
                           output=['rgba', 'dpeth', 'semantic'],
                           )

### Output:
[15:46:22:209882]:[Warning]:[Assets] ResourceManager.cpp(353)::loadSemanticSceneDescriptor : SSD File Naming Issue! Neither SceneInstanceAttributes-provided name : `data/scene_datasets/habitat-test-scenes/apartment_1.scn` nor constructed filename : `data/scene_datasets/habitat-test-scenes/info_semantic.json` exist on disk.
[15:46:23:262974]:[Warning]:[Sim] Simulator.cpp(508)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0
Traceback (most recent call last):
  File "/home/solomon/workspace/playground/extract_image.py", line 114, in <module>
    extractor = ImageExtractor(scene_file, 
  File "/home/solomon/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.2.5-py3.9-linux-x86_64.egg/habitat_sim/utils/data/data_extractor.py", line 122, in __init__
    self.poses = self.pose_extractor.extract_all_poses()
  File "/home/solomon/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.2.5-py3.9-linux-x86_64.egg/habitat_sim/utils/data/pose_extractor.py", line 63, in extract_all_poses
    return np.array(poses)
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 2 dimensions. The detected shape was (603, 3) + inhomogeneous part.

solomon-ma, 2023-09-18:
already solved by use v0.2.1 habitat_sim

closed 2023-09-18 by solomon-ma

vcozzolino, 2024-02-07:
Just faced the same problem with `v0.3.0` when using as scene the example from MatterPort3D that you can download following the tutorial (`data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.glb`).

I had a look at the source code, and fixed it by editing the function `extract_all_poses` in `habitat_sim/utils/data/pose_extractor.py`. I basically replaced the return value to `np.array(poses, dtype=object)` as the `poses` contain heterogeneous data types.