facebookresearch/habitat-sim #2210: Module not found error

SarthakJaindebugger, 2023-09-23:
I'm facing the same issue too after running this command: python examples/viewer.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
I installed habitat-sim and lab via this link: https://github.com/facebookresearch/sound-spaces/issues/80
I tried it on both Ubuntu and Google Colab and I've a system with built in display.

_Originally posted by @SarthakJaindebugger in https://github.com/facebookresearch/habitat-sim/issues/1702#issuecomment-1732270634_

aclegg3, 2023-09-25:
Hey @SarthakJaindebugger 

In that Issue, the recommended install for habitat-sim looks like:

```
python setup.py install --with-cuda --bullet --audio --headless # compilation goes brrrr...
```

If you install with `--headless` you will not be able to run the viewer application.

SarthakJaindebugger, 2023-09-25:
Thanks @aclegg3 for the reply. I'll try by running this command by removing --headless and --cuda. As I don't have a GPU on my system and it's a laptop as well.

aclegg3, 2024-01-16:
Closing stale issue.

closed 2024-01-16 by aclegg3