facebookresearch/habitat-sim #2213: "TypeError: cannot pickle '_magnum.Color4' object" when running benchmark.py in v0.2.5

sanjayharesh, 2023-09-26:
Hi there,

I'm trying to run the benchmarking script [here](https://github.com/facebookresearch/habitat-sim/blob/v0.2.5/examples/benchmark.py) in v0.2.5 and I'm getting the following error. 

`TypeError: cannot pickle '_magnum.Color4' object`

Unfortunately, don't get more detail in the error message thought it seems to be because of multiprocessing and removing the multiprocessing here: https://github.com/facebookresearch/habitat-sim/blob/c8887c8cf421b6df2d77489d42f4e20488e184eb/examples/demo_runner.py#L391

seems to make it work. 

For completeness, here's my setup
habitat-sim v0.2.5 installed from source. 
Ubuntu 18.04
Python 3.9 and Cuda version 11.7.

aclegg3, 2023-09-27:
Thanks for raising this, @sanjayharesh,

I can reproduce this issue locally.

Looks like this particular code isn't run our CI, so possible it has become stale.

mosra, 2023-10-26:
FYI, this should be implemented with https://github.com/mosra/magnum-bindings/commit/f561337caa9094c4a99411fac402342343f4ecb4, however I don't have an easy way to verify that it indeed works in the full context of Habitat. 

Assuming #2243 is included in your working copy, it should be enough to just pull the submodule in `src/deps/magnum-bindings` to this commit to get the change and check.

ORionTMT, 2023-11-08:
I also ran into the same problem with v0.3.0, have you managed to solve it yet?