facebookresearch/habitat-sim #2216: Unable to create windowless context

MateusLima2801, 2023-09-27:
I booted Linux in my original Windows system and I'm trying to run the examples of Habitat-Lab, but the output is constantly:

```
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
WindowlessContext: Unable to create windowless context
```

I ran` $ hwinfo --gfxcard --short` to check the graphics card and got:

```
graphics card:                                                  
                       Intel UHD Graphics 620

Primary display adapter: #19
```
Do you, guys, know if I have to use NVIDIA GPUs instead of my current graphic card?

MateusLima2801, 2023-09-27:
I installed habitat-sim through:

`conda install habitat-sim withbullet -c conda-forge -c aihabitat`

I didn't use "--headless". I am trying to run the scripts in habitat-lab/examples/interactive_play.py through the command:

`python interactive_play.py --never-end`

We want to use Habitat for research, so it would be nice to know if it's necessary to have NVIDIA GPU for running the simulations.

closed 2023-10-04 by MateusLima2801

arjo129, 2024-01-13:
What was the solution you found? I've run into a similar issue. I can run the viewer though.