facebookresearch/habitat-sim #2226: Problem loading notebook ECCV 2020: Navigation in colab

digiamm, 2023-10-03:
Hi guys,

I am simply trying to execute the navigation ECCV 2020: Navigation in Colab. From the beginning, running the first cell:

```
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2732  100  2732    0     0  12605      0 --:--:-- --:--:-- --:--:-- 12648
--2023-10-03 16:45:56--  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.continuum.io (repo.continuum.io)... 104.16.246.55, 104.16.245.55, 2606:4700::6810:f537, ...
Connecting to repo.continuum.io (repo.continuum.io)|104.16.246.55|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh [following]
--2023-10-03 16:45:56--  https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.anaconda.com (repo.anaconda.com)... 104.16.130.3, 104.16.131.3, 2606:4700::6810:8303, ...
Connecting to repo.anaconda.com (repo.anaconda.com)|104.16.130.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 103219356 (98M) [application/x-sh]
Saving to: ‘Miniconda3-latest-Linux-x86_64.sh’

Miniconda3-latest-L 100%[===================>]  98.44M   165MB/s    in 0.6s    

2023-10-03 16:45:57 (165 MB/s) - ‘Miniconda3-latest-Linux-x86_64.sh’ saved [103219356/103219356]

PREFIX=/usr/local
Unpacking payload ...
                                                                                 
Installing base environment...

Downloading and Extracting Packages

Downloading and Extracting Packages

Preparing transaction: done
Executing transaction: done
installation finished.
WARNING:
    You currently have a PYTHONPATH environment variable set. This may cause
    unexpected behavior when running the Python interpreter in Miniconda3.
    For best results, please verify that your PYTHONPATH only points to
    directories of packages that are compatible with the Python interpreter
    in Miniconda3: /usr/local
Collecting package metadata (current_repodata.json): done
Solving environment: unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: | 
```
The `Solving environment` is endless. I don't know, seems to be an issue related to Conda. Do you have any clue?

aclegg3, 2023-10-03:
@digiamm we are aware that Colab notebook install is not functional presently. Sorry for the inconvenience. Please install and run the code locally for the time being. :)

digiamm, 2023-10-04:
Ok, thanks. When I move locally, everything is fine with the conda installation, at least according to your root README. I can test the installation with the viewer as you propose by running:
`python examples/viewer.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`
All good till now.  However, when I try to execute the Jupiter notebook locally I have an issue related to `habitat` package:

```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
/local/home/path-to-habitat/habitat-sim/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb Cell 4 line 5
     54 # import the maps module alone for topdown mapping
     55 if display:
---> 56     from habitat.utils.visualizations import maps

ModuleNotFoundError: No module named 'habitat'
```
Apparently `habitat_sim` package is fine but `habitat` no. Could you help me with this?

aclegg3, 2023-10-04:
@digiamm did you install [habitat-lab](https://github.com/facebookresearch/habitat-lab)? That code is importing some mapping utils from lab.

digiamm, 2023-10-05:
Thank you! I was missing `habitat-lab`.

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead.

closed 2024-01-16 by aclegg3