facebookresearch/habitat-sim #2229: Is headless mode installed by default?

MateusLima2801, 2023-10-04:
Hi, there, 

I'm running in Linux but I don't have NVIDIA GPUs for use in headless mode, as described necessary in [this issue](https://github.com/facebookresearch/habitat-sim/issues/1511).

I installed habitat-sim using:

`conda install habitat-sim -c conda-forge -c aihabitat
`

I've downloaded the datasets and I was trying to run the navigation example in [the Quickstart Page](https://aihabitat.org/docs/habitat-lab/quickstart.html) and I ran into a problem with CUDA, below it follows all the output:

```
pybullet build time: Sep 25 2023 15:30:28
2023-10-04 10:56:32,022 Initializing dataset PointNav-v1
2023-10-04 10:56:44,264 initializing sim Sim-v0
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
WindowlessContext: Unable to create windowless context
```

Anyone can think of an explanation/a way to solve it?

aclegg3, 2023-10-09:
Hey @MateusLima2801,

The version you linked should be a display mode, not headless. I've seen these issues before, they usually point to a driver or config issue locally.

@0mdc may also have some thoughts if you provide more system details if you want debug farther.

0mdc, 2023-10-09:
Hey @MateusLima2801 

Can you try using the environment variable `MAGNUM_LOG=verbose` when running Habitat? This may give additional hints.

>I've seen these issues before, they usually point to a driver or config issue locally.

This corroborates with my experience. I've seen this issue occur a few times due to an invalid `glvnd` installation. I suspect that this could also occur if you're using a laptop with both an integrated and dedicated GPU, and the driver misbehaves.

MateusLima2801, 2023-10-10:
Hi, guys, thanks for the feedback. I've set the environment variables through:

`export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON`

And what I had in return was:

```2023-10-05 16:04:16,125 Initializing dataset PointNav-v1
2023-10-05 16:04:16,419 initializing sim Sim-v0
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_EXT_device_drm EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=1): EGL_MESA_device_software EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
```
My GPU is "_Intel UHD Graphics 620_" and I'm running on _Ubuntu 23.04_.

0mdc, 2023-10-10:
Can you check if [glvnd](https://github.com/NVIDIA/libglvnd) is installed on your machine? This library may fix your issue.

MateusLima2801, 2023-10-11:
I followed instructions on [glvnd](https://github.com/NVIDIA/libglvnd) installation and installed `libglvnd-dev`, but it was unsuccessfull over this issue. Running this [Habitat script](https://aihabitat.org/docs/habitat-lab/quickstart.html) still gives the same output:

```
2023-10-11 10:46:51,623 Initializing dataset PointNav-v1
2023-10-11 10:46:51,932 initializing sim Sim-v0
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_EXT_device_drm EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=1): EGL_MESA_device_software EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
WindowlessContext: Unable to create windowless context
```

@0mdc, any more ideas?

0mdc, 2023-10-25:
Hey @MateusLima2801,

Apologize for the late reply. Have you managed to find a solution?