facebookresearch/habitat-sim #2240: Build error using command `python setup.py install --headless --audio` on branch RLRAudioPropagationUpdate

EntyccE, 2023-10-14:
I'm installing soundspace 2.0 following the instruction [https://github.com/facebookresearch/sound-spaces/blob/main/INSTALLATION.md](url) and I encountered the following error. In brief, cmake cannot found RLRAudioPropagation_LIBRARY variable, which is 
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
RLRAudioPropagation_LIBRARY
    linked by target "sensor" in directory D:/Programmes/soundspace/habitat-sim/src/esp/sensor
```
I'm running on a Windows system and my cmake version is the latest. Though the instruction seems to install a version of 3.14.0 but after I run the command of creating the conda environment, it's said that cmake cannot be found. 

The whole output of command `python setup.py install --headless --audio` is 
```
D:\Programmes\soundspace\habitat-sim\setup.py:22: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.util import strtobool
running install
running bdist_egg
running egg_info
writing src_python\habitat_sim.egg-info\PKG-INFO
writing dependency_links to src_python\habitat_sim.egg-info\dependency_links.txt
writing requirements to src_python\habitat_sim.egg-info\requires.txt
writing top-level names to src_python\habitat_sim.egg-info\top_level.txt
reading manifest file 'src_python\habitat_sim.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'src_python\habitat_sim.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
Using relative rpath
CMake Deprecation Warning at deps/corrade/CMakeLists.txt:31 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at deps/assimp/CMakeLists.txt:39 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Shared libraries disabled
-- Looking for DirectX...
-- DirectX_PREFIX_PATH changed.
-- Could not locate DirectX
-- Looking for ZLIB...
-- Checking for module 'zzip-zlib-config'
--   No package 'zzip-zlib-config' found
-- Found ZLIB: optimized;D:/anaconda3/Library/lib/z.lib;debug;D:/anaconda3/Library/lib/z.lib
-- Build an import-only version of Assimp.
CMake Deprecation Warning at deps/assimp/code/CMakeLists.txt:46 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning (dev) at C:/Users/whao_e/AppData/Local/Programs/Python/Python310/Lib/site-packages/cmake/data/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (rt) does
  not match the name of the calling package (RT).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  deps/assimp/cmake-modules/FindRT.cmake:19 (find_package_handle_standard_args)
  deps/assimp/code/CMakeLists.txt:1013 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find rt (missing: RT_LIBRARY)
[... 149 lines cut ...]
    cmd_obj.run()
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\site-packages\setuptools\command\bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\site-packages\setuptools\command\bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\distutils\dist.py", line 985, in run_command        
    cmd_obj.run()
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\site-packages\setuptools\command\install_lib.py", line 11, in run
    self.build()
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\distutils\command\install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\distutils\dist.py", line 985, in run_command        
    cmd_obj.run()
  File "D:\Programmes\soundspace\habitat-sim\setup.py", line 271, in run
    self.build_extension(ext)
  File "D:\Programmes\soundspace\habitat-sim\setup.py", line 372, in build_extension
    subprocess.check_call(
  File "C:\Users\whao_e\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\whao_e\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\cmake\\data\\bin\\cmake', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=D:\\Programmes\\soundspace\\habitat-sim\\build\\lib.win-amd64-3.10\\habitat_sim\\_ext', '-DPYTHON_EXECUTABLE=C:\\Users\\whao_e\\AppData\\Local\\Programs\\Python\\Python310\\python.exe', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DREL_BUILD_RPATH=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_WITH_VHACD=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF', '-DBUILD_BASIS_COMPRESSOR=OFF', '-DBUILD_WITH_AUDIO=ON', 'D:\\Programmes\\soundspace\\habitat-sim\\src']' returned non-zero exit status 1.
```

EntyccE, 2023-10-14:
Oh it seem that Windows is not suitable to build that. Linux is fine.

EntyccE, 2023-11-12:
I tried on a linux one and it also generate this error. can someone help to diagnose where is the reason?

Shelton-wong, 2025-05-27:
> I tried on a linux one and it also generate this error. can someone help to diagnose where is the reason?

I also encounter this error，have you solved this？could you tell me how solve this？