facebookresearch/habitat-sim #2248: Assertion Error after adding static object and recompute navmesh from Viewer.py

EvaLiiiiii, 2023-10-20:
So I'm leveraging this tutorial for [adding object](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb#scrollTo=asWjR77S1Ufe&uniqifier=2) "Clutter Generation Script" and the viewer.py example. I basically followed what's in the tutorial for adding object and then try to use the navmesh_config_and_recompute() provided in viewer.py so the object can become obstacles in the navmesh but it's keep failing on an  assertion error  
[image]
Does anyone have any insights on what this function does? or where the error may come from? 
After adding the object and setting self.navmesh_settings.include_static_objects = True, when I call the sim.recompute_navmesh, this error shows up.. Thanks!

aclegg3, 2023-10-20:
@EvaLiiiiii so strange. :thinking: 
My guess would be some edge case related to the object asset mesh import. That assertion happens when the `ResourceManager::getMeshMetaData` function cannot find a meshMetaData entry for the object. Mesh data is queried to join all meshes in the scene before constructing the navmesh.

If you can provide the object asset I could test it locally and attempt to debug.
If not, you could try passing the mesh asset through a converter to another format. That may clean-up issues or bypass any importer bug. Habitat importers prefer glTF.

Also, what version are you on? I see a conda build here, so it could be that if you are not on nightly the bug is stale and resolved elsewhere upstream.

EvaLiiiiii, 2023-10-20:
Hi @aclegg3 , Thanks for quick response! I'm working on the "stable" branch. I'm testing with sphere object in Adrian.glb from Gibson.
[image]
This is what I did. I've also changed "sample_object_state" function to be sampling near a point instead of complete random sample. The spheres seem to appear correctly. But when I press shift+ n to recompute the navmesh I get that error..

EvaLiiiiii, 2023-10-20:
Hi @aclegg3  I think the problem is with the sphere object. After I changed it to the donut it worked!

aclegg3, 2023-10-20:
Yeah, must be an asset issue with the sphere. Glad you have a workaround and thanks for the report.
When we get some bandwidth we can dig into why this is crashing.