facebookresearch/habitat-sim #2252: Plugins missing when using Habitat sim via the C++ interfaces

benborder, 2023-10-26:
## Habitat-Sim version
v0.2.5

## ❓ Questions and Help
I am using Habitat sim via the C++ interfaces. However when calling the `esp::sim::Simulator` constructor, an exception is thrown:
```
PluginManager::Manager::setPreferredPlugins(): ObjImporter is not a known alias
```

It seems the plugins are statically compiled, but when inspecting the `AbstractManager`, the `_state->plugins` and `_state->aliases` are empty. The `AssimpImporter` plugin is built from `MagnumPlugins` and `importStaticPlugin.cpp` is compiled and from what I can tell linked correctly. So static plugins should automatically be registering themselves, but for some reason they are not.

Am I missing something or not initialising something correctly?

For reference I have added habitat sim to my C++ project via cmake FetchContent:

```cmake
include(FetchContent)
FetchContent_Declare(
	habitat_sim
	GIT_REPOSITORY https://github.com/facebookresearch/habitat-sim.git
	GIT_TAG        v0.2.5
)
set(BUILD_PYTHON_BINDINGS OFF)
set(BUILD_DATATOOL OFF)
set(BUILD_GUI_VIEWERS OFF)
set(BUILD_WITH_BULLET ON)
set(BUILD_WEB_APPS OFF)
set(BUILD_WITH_BACKGROUND_RENDERER ON)
set(BUILD_WITH_CUDA OFF)
set(MAGNUM_BUILD_PLUGINS_STATIC ON) # not sure if this is required, tried it with and without
set(CMAKE_POLICY_DEFAULT_CMP0072 NEW) # use glvnd if available
set(CMAKE_POLICY_DEFAULT_CMP0077 NEW) # dont ignore variables already set when a new option is defined
FetchContent_Populate(habitat_sim)
add_subdirectory(${habitat_sim_SOURCE_DIR}/src ${habitat_sim_BINARY_DIR} EXCLUDE_FROM_ALL)

add_library(test_lib STATIC
  src/test.cpp
)

target_link_libraries(test_lib
PUBLIC
  Threads::Threads
PRIVATE
  $<BUILD_INTERFACE:agent>
  $<BUILD_INTERFACE:assets>
  $<BUILD_INTERFACE:assimp>
  $<BUILD_INTERFACE:scene>
  $<BUILD_INTERFACE:core>
  $<BUILD_INTERFACE:geo>
  $<BUILD_INTERFACE:gfx>
  $<BUILD_INTERFACE:metadata>
  $<BUILD_INTERFACE:io>
  $<BUILD_INTERFACE:nav>
  $<BUILD_INTERFACE:sensor>
  $<BUILD_INTERFACE:physics>
  $<BUILD_INTERFACE:sim>
  $<BUILD_INTERFACE:MagnumPlugins::AssimpImporter> # tried with and without
)

add_executable(test
  main.cpp
)

target_link_libraries(test
PUBLIC
  test_lib
)
```
I use the following code in test.cpp:

```cpp
static esp::logging::LoggingContext g_loggin_context;

void test_sim() {
  esp::sim::SimulatorConfiguration sim_config;
  sim_config.enablePhysics = true;
  sim_ = std::make_unique<esp::sim::Simulator>(sim_config); // <- throws here
  // do stuff
}
```
The call stack when the exception is thrown is as follows:
```
libc.so.6!pthread_kill (Unknown Source:0)
libc.so.6!raise (Unknown Source:0)
libc.so.6!abort (Unknown Source:0)
Corrade::PluginManager::AbstractManager::setPreferredPlugins(Corrade::PluginManager::AbstractManager * const this, const Corrade::Containers::StringView alias, const Corrade::Containers::StringIterable & plugins) (/home/dev/test/build/Debug/_deps/habitat_sim-src/src/deps/corrade/src/Corrade/PluginManager/AbstractManager.cpp:484)
esp::assets::ResourceManager::buildImporters(esp::assets::ResourceManager * const this) (/home/dev/test/build/Debug/_deps/habitat_sim-src/src/esp/assets/ResourceManager.cpp:125)
esp::assets::ResourceManager::ResourceManager(esp::assets::ResourceManager * const this, esp::metadata::MetadataMediator::ptr _metadataMediator) (/home/dev/test/build/Debug/_deps/habitat_sim-src/src/esp/assets/ResourceManager.cpp:117)
std::make_unique<esp::assets::ResourceManager, std::shared_ptr<esp::metadata::MetadataMediator>&>() (/usr/include/c++/11/bits/unique_ptr.h:962)
esp::sim::Simulator::reconfigure(esp::sim::Simulator * const this, const esp::sim::SimulatorConfiguration & cfg) (/home/dev/test/build/Debug/_deps/habitat_sim-src/src/esp/sim/Simulator.cpp:115)
esp::sim::Simulator::Simulator(esp::sim::Simulator * const this, const esp::sim::SimulatorConfiguration & cfg, esp::metadata::MetadataMediator::ptr _metadataMediator) (/home/dev/test/build/Debug/_deps/habitat_sim-src/src/esp/sim/Simulator.cpp:56)
std::make_unique<esp::sim::Simulator, esp::sim::SimulatorConfiguration&>() (/usr/include/c++/11/bits/unique_ptr.h:962)
```

mosra, 2023-10-26:
Can you try linking the plugin to the final executable instead of the library, or putting it to an `INTERFACE` section instead of `PRIVATE`? That should work I think.

benborder, 2023-10-27:
Yep, adding the plugins as `INTERFACE` in target_link_libraries for `test_lib` worked. Thanks for the help!

However, now `importStaticPlugin.cpp` is getting compiled 4 times. Once in `magnum-plugins`, again in `assets` (or other habitat based lib), then in my own code `test_lib` and finally in `test`. Its not really an issue, but doesn't seem ideal.

closed 2023-10-27 by benborder

mosra, 2023-10-27:
Yeah, that's known -- it's tracked as an internal tech debt issue and will be fixed eventually. 

In short, basically none of the intermediate static libraries should link to any plugins, only the final executables or shared libraries.