facebookresearch/habitat-sim #2254: Unable to download hssd dataset

taylorlulu, 2023-11-01:
When I'm trying to download hssd dataset using the command 
`python -m habitat_sim.utils.datasets_download --uids hssd-hab --data-path data/`
It reported 
`git clone --depth 1 --branch main git@hf.co:datasets/hssd/hssd-hab.git /home/lu/Desktop/embodied_ai/data/versioned_data/hssd-hab
Cloning into '/home/lu/Desktop/embodied_ai/data/versioned_data/hssd-hab'...
/home/lu/.ssh/config: line 6: Bad configuration option: password
/home/lu/.ssh/config: terminating, 1 bad configuration options
fatal: Could not read from remote repository.`
When I open the link with the browser 
[image]
How can I download the hssd dataset?

Jinghan11, 2023-11-09:
The appropriate link is https://huggingface.co/datasets/hssd/hssd-hab.

However, I'm encountering issues when trying to download this dataset. While I can successfully execute the command` git clone https://huggingface.co/datasets/hssd/hssd-hab`, I don't seem to have and don't konw how to get the necessary permissions to use` python -m habitat_sim.utils.datasets_download --uids hssd-hab` for downloading the dataset. This issue could potentially be due to the datasets_download script using SSH instead of HTTPS.

taylorlulu, 2023-11-09:
@Jinghan11 Yes, I successfully run the `git clone https://huggingface.co/datasets/hssd/hssd-hab` command as well. I agree it may be something wrong with the `habitat_sim.utils.datasets_download`.

closed 2023-11-09 by taylorlulu

Jinghan11, 2023-11-09:
Have you ever tried running a program related to Habitat-3.0 Multi-Agent Training? I am not sure if this problem is causing the FileNotFoundError when I run `python habitat_baselines/run.py --config-name=social_nav/social_nav.yaml`. I have opened an [issue](https://github.com/facebookresearch/habitat-lab/issues/1677) about this in the habitat-lab repo.

taylorlulu, 2023-11-15:
@Jinghan11 No I haven't. I'll try it.