facebookresearch/habitat-sim #2258: Determine the visual overlap between two cameras?

janblumenkamp, 2023-11-03:
I would like to determine the visual overlap between two cameras on two independent agents. Is there any way to do this from within habitat?
Thanks!

0mdc, 2023-11-03:
Hello @janblumenkamp,

What exactly do you mean by visual overlap? Which type of sensor are you working with? What kind of output are you looking for?

janblumenkamp, 2023-11-04:
I am thinking of any kind of camera, e.g. RGB, depth or semantic. Specifically, I would like to know how much of a scene observed through one camera is also observed through a second camera.

In the simplest case, I would like to know what percentage of two viewing frustum overlaps. This is a trivial calculation given the camera extrinsic and intrinsic, assuming a plane. Realistically though, there are occlusions, and the same scene viewed by two different cameras might have a smaller visual overlap.

Essentially, it would probably require casting a ray through each pixel of camera A and checking for each pixel whether a ray can also be cast by camera B without hitting any object in the scene. Does that make sense?

0mdc, 2023-11-06:
That makes sense.

I believe that you should be able to calculate this using Python.

To get a ray from a specific viewport point, you can use `RenderCamera.unproject()` ([ref](https://github.com/facebookresearch/habitat-sim/blob/main/src/esp/gfx/RenderCamera.h#L240)). You can then use physics raycasting (`Simulator.cast_ray()`). Note that this requires the Bullet build.

In [this example viewer](https://github.com/facebookresearch/habitat-sim/blob/main/examples/viewer.py#L690), the handling of mouse clicks is done using this method. In your case, you would sample the sensor viewport iteratively instead of using the mouse position.

After that, I suppose that the hit positions of the compared sensors would need to be stored in some spatial data structure so that the visual overlap can be processed.

@aclegg3 Thoughts?