facebookresearch/habitat-sim #2263: ImportError: cannot import name 'shaders' from 'magnum' for python examples/viewer.py

satyapalsinh10, 2023-11-12:
Dear team,

I am using
## Habitat-Sim version
main

I am trying to execute `python examples/viewer.py` using python 3.9.18 and cmake=3.14.0.

I did install habitat-sim with physics on headless machines using conda.

while executing I am facing the below error:  

## 🐛 Bug

` Traceback (most recent call last):
  File "/scratch/smg8252/environments/habitat-sim/examples/viewer.py", line 22, in <module>
    from magnum import shaders, text
ImportError: cannot import name 'shaders' from 'magnum' (/home/smg8252/.local/lib/python3.9/site-packages/magnum/__init__.py)
`
## Steps to Reproduce

Steps to reproduce the behavior:

1. git clone https://github.com/facebookresearch/habitat-sim.git
2. conda create -n habitat python=3.9 cmake=3.14.0
3. conda activate habitat
4. conda install habitat-sim withbullet headless -c conda-forge -c aihabitat
5. python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path /path/to/data/
6. python -m habitat_sim.utils.datasets_download --uids habitat_example_objects --data-path /path/to/data/
7. export PYTHONPATH=$(pwd)  #inside 'habitat-sim/' directory
8. python examples/viewer.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

Can you please help ??

Thank you

nikhilanj, 2023-12-04:
https://github.com/facebookresearch/habitat-sim/issues/2049#issuecomment-1490568606

This solved it for me

chenhaomingbob, 2024-06-15:
Hi.
In versions 0.3.1 & 0.2.4, I encourage the same error. 
Did you solve this error?