facebookresearch/habitat-sim #2269: EGL BAD PARAMETER

edwardjjj, 2023-11-24:
## Habitat-Sim version
Habitat-Challenge-2022

## ❓ Questions and Help
Hello everyone. I'm trying to reproduce a paper called l3mvn using docker. Originally I tried to use the cudagl 10.1 image, but it did not support 4090 graphics card. After a bit of searching, I switched to cudagl:11.7.1 but encountered the following error message. Does anyone have solutions working with newer generation cards? Thank you very much in advance!

### Environment
OS: Ubuntu 20.04-wsl on Windows 11
GPU: 4090

### Error message
[image]

### Dockerfile

```
FROM fairembodied/cudagl:11.7.1-cudnn8-devel-1.2-glvnd-devel-ubuntu20.04
ARG MYENV=habitat

RUN apt-get update && apt-get install -y --no-install-recommends build-essential git curl vim \
  ca-certificates libjpeg-dev libglm-dev libegl1-mesa-dev xorg-dev freeglut3-dev pkg-config wget zip unzip openssh-client \
  && rm -rf /var/lib/apt/lists/*

RUN curl -L -o ~/miniconda.sh -O  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh  &&\
    chmod +x ~/miniconda.sh &&\
    ~/miniconda.sh -b -p /opt/conda &&\
    rm ~/miniconda.sh &&\
    /opt/conda/bin/conda install numpy pyyaml scipy ipython mkl mkl-include &&\
    /opt/conda/bin/conda clean -ya

ENV PATH="/opt/conda/bin:/usr/local/nvidia/bin:/usr/local/cuda/bin:/usr/local/sbin:\
  /usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

RUN mkdir -p /home/app

RUN conda create -n $MYENV python=3.7 cmake=3.14.0
RUN conda install -n $MYENV pytorch==1.13.1 torchvision==0.14.1 torchaudio==0.13.1 pytorch-cuda=11.7 -c pytorch -c nvidia
RUN conda install habitat-sim-challenge-2022 headless -c conda-forge -c aihabitat -n $MYENV
RUN git clone --branch challenge-2022 https://github.com/facebookresearch/habitat-lab.git /home/app/habitat-lab
RUN conda run -n $MYENV /bin/bash -c \
  "cd /home/app/habitat-lab; pip install -e ."
RUN conda run -n $MYENV /bin/bash -c \
  "python -m pip install 'git+https://github.com/facebookresearch/detectron2.git@v0.6'"

WORKDIR /home/app
COPY ./requirements.txt /home/app
RUN conda run -n $MYENV /bin/bash -c \
  "pip install -r requirements.txt"

WORKDIR /home/app/l3mvn
```
###nvidia-smi output
[image]

0mdc, 2023-11-24:
Hey @edwardjjj,

The latest GPUs should work just fine.
This is most likely a configuration issue as some level (drivers, libraries (EGL), wsl/docker).

Some quick troubleshooting steps:
* What command did you use to get the error?
* Can you run Habitat with the following environment variables: `MAGNUM_LOG=verbose`, `MAGNUM_GPU_VALIDATION=ON`? The program output will indicate additional information before the error.
* Does `eglinfo` indicate a problem?
* Are you running on X11 or Wayland? Can you try the other?

A similar issue was encountered on a similar environment in [this thread](https://github.com/facebookresearch/habitat-sim/issues/978). This probably will help troubleshooting.

[This thread](https://stackoverflow.com/questions/63294883/opengl-egl-returns-0x300c-egl-bad-parameter-at-eglgetdisplay-in-docker) may also apply to you, as you are building from a similar environment. Make sure to delete your `build` folder after you fiddle with system libraries so that you don't pull out the issue from cache when rebuilding.

edwardjjj, 2023-11-25:
Hi @0mdc,

Thank you for your prompt response. I reverted back to the **cudagl:10.1-devel-ubuntu18.04** image and ran the container. 
## `eglinfo` on the host
```
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base
    EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions
    EGL_KHR_debug EGL_KHR_platform_x11 EGL_EXT_platform_x11
    EGL_EXT_platform_device EGL_MESA_platform_surfaceless
    EGL_EXT_explicit_device EGL_KHR_platform_gbm EGL_MESA_platform_gbm

GBM platform:
eglinfo: eglInitialize failed

X11 platform:
eglinfo: eglInitialize failed

Device platform:
eglinfo: eglInitialize failed
```
## The `eglinfo` inside the container:
```
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base EGL_EXT_device_enumeration
    EGL_EXT_device_query EGL_KHR_client_get_all_proc_addresses
    EGL_EXT_client_extensions EGL_KHR_debug EGL_KHR_platform_x11
    EGL_EXT_platform_x11 EGL_EXT_platform_device
    EGL_MESA_platform_surfaceless EGL_EXT_explicit_device
    EGL_KHR_platform_gbm EGL_MESA_platform_gbm EGL_EXT_platform_wayland

Segmentation fault
```
### I'm confused here. So does this mean the container depends on the host EGL environment to function?
## `ldconfig -p | grep libEGL` output:
```
libEGL_nvidia.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
        libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
        libEGL_mesa.so.0 (libc6) => /usr/lib/i386-linux-gnu/libEGL_mesa.so.0
        libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
        libEGL.so.1 (libc6) => /usr/lib/i386-linux-gnu/libEGL.so.1
        libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
        libEGL.so (libc6) => /usr/lib/i386-linux-gnu/libEGL.so
```

## Full output with `MAGNUM_LOG=verbose` and `MAGNUM_GPU_VALIDATION=ON`:
```
2023-11-25 00:57:09,741 Initializing dataset ObjectNav-v1
Scenes per thread:
Thread 0: ['4ok3usBNeis', '5cdEh9F2hJL', '6s7QHgap2fW', 'DYehNKdT76V', 'Dd4bFSTQ8gi', 'Nfvxx8J5NCo', 'QaLdnwvtxbs', 'TEEsavR23oF', 'XB4GS9ShBRE', 'bxsVRursffK', 'cvZr5TUy5C5', 'mL8ThkuaVTM', 'mv2HUxq3B53', 'p53SfW6mjZe', 'q3zU7Yy5E5s', 'qyAac8rV8Zk', 'svBbv1Pavdk', 'wcojb4TFT35', 'ziup5kvtCCR', 'zt1RVoi7PcG']
2023-11-25 00:57:12,258 Initializing dataset ObjectNav-v1
2023-11-25 00:57:13,067 initializing sim Sim-v0
[00:57:13:078395]:[Metadata] AttributesManagerBase.h(352)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : default.scene_dataset_config.json from original name : default| This file does not exist.
[00:57:13:078483]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DSolid:capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
[00:57:13:078508]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DWireframe:capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
[00:57:13:078526]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneSolid:coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
[00:57:13:078541]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneWireframe:coneWireframe_segments_32_halfLen_1.25) created and registered.
[00:57:13:078556]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeSolid:cubeSolid) created and registered.
[00:57:13:078570]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeWireframe:cubeWireframe) created and registered.
[00:57:13:078587]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderSolid:cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
[00:57:13:078602]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderWireframe:cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
[00:57:13:078618]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereSolid:icosphereSolid_subdivs_1) created and registered.
[00:57:13:078634]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereWireframe:icosphereWireframe_subdivs_1) created and registered.
[00:57:13:078649]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereSolid:uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
[00:57:13:078660]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereWireframe:uvSphereWireframe_rings_16_segments_32) created and registered.
[00:57:13:078668]:[Metadata] AssetAttributesManager.cpp(110)::AssetAttributesManager : Built default primitive asset templates : 12
[00:57:13:079062]:[Metadata] SceneDatasetAttributesManager.cpp(35)::createObject : File (default) not found, so new default dataset attributes created  and registered.
[00:57:13:079075]:[Metadata] MetadataMediator.cpp(120)::createSceneDataset : Dataset default successfully created.
[00:57:13:079086]:[Metadata] AttributesManagerBase.h(352)::createFromJsonOrDefaultInternal : <Physics Manager>: Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json| This file does not exist.
[... cut ...]
  File "/home/app/envs/habitat/utils/vector_env.py", line 416, in close
    write_fn((CLOSE_COMMAND, None))
  File "/opt/conda/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/opt/conda/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/opt/conda/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
```

edwardjjj, 2023-12-20:
Seems like a WSL specific issue, closing here.

closed 2023-12-20 by edwardjjj