facebookresearch/habitat-sim #2271: The dataset in hugging face seems to be not available?

Ethan2415, 2023-11-27:
## Habitat-Sim 0.3.0(latest)

python -m habitat_sim.utils.datasets_download --uids habitat_test_pointnav_dataset --data-path data/
seems to not be working, and so does other datasets? When i enter the address attached in the datasets_download.py file, it show 404. 
Where else can I download those dataset?

aclegg3, 2023-11-27:
Hey @Ethan2415,

Sometimes network issues prevent access to AWS S3 (in this case) or HuggingFace servers temporarily. This and other data sources are working for me presently. Please try again and update this issue.

Lr-2002, 2024-02-03:
@Ethan2415  hello,
Were you using your web-browser to view the website(something.git) ? The view with ".git" is not accessable for me both WHILE you can use git to clone it. This might because of the huggingface web.

aclegg3, 2024-02-05:
Outages happen. Closing this task as stale and likely not a bug.

closed 2024-02-05 by aclegg3