facebookresearch/habitat-sim #2302: No module named 'habitat_sim.utils.settings'

Thaumaturge2020, 2024-01-19:
## Habitat-Sim version
vx.x.x or main?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? The bug you are about to report may already be fixed in the latest version.

Main branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1.export PYTHONPATH=${pwd}
2.python examples/example.py --scene ./test_data/scene_datasets/habitat-test-scenes/skokloster-castle.glb --enable_physics
3.

I'm running the command right in my folder

 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/main/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python src_python/habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

aclegg3, 2024-01-19:
Hey @Thaumaturge2020,

Looks like the settings module is not being found. I'm curious what version you installed and how. Did you conda install or build from source?
I see the `--enable-physics` flag in your command which is no longer part of the viewer CLI on `main` branch. This makes me think you may be on an older version of the repo.

AbhinandanVellanki, 2025-11-18:
hello @aclegg3 

I have installed with conda and I would like to run the simulator. I cant run the commands with "python -m ... " because I get the same ModuleNotFoundError. 

Can you please describe how to run the simulator in interactive mode after installing using conda?