facebookresearch/habitat-sim #2310: A100 Render Problem: Black Squares

maytusp, 2024-01-29:
## Habitat-Sim version
v0.3.0 

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/
Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes

Perhaps your question is answered there. If not, carry on!
Yes

## ❓ Questions and Help

 
I’ve found an issue on all A100 GPUs.
It has black squares on every frame that the simulation render.
 
This problem never occurred with V100 GPUs, so I guess it’s the A100 specific problem.
I also found that the OpenGL software versions are the same for both V100 and A100. 
I’m not sure how to fix this issue. It seems I must change rendering setting.
 
[image]

[image]

aclegg3, 2024-02-29:
hey @maytusp,

While I have not observed this issue personally, it seems to me likely that driver support is culprit. 

I don't really know how to debug this further except to try re-installing/updating the gl and cuda drivers on the system. 

@0mdc have we seen this anywhere? I think we've used A100s without issue. :thinking:

0mdc, 2024-02-29:
I don't believe that we've seen this issue.

This may be a hardware/driver-specific issue for A100s: https://forums.developer.nvidia.com/t/black-artifacts-when-rendering-pixel-obs-on-a100-gpus/255540

maytusp, 2024-03-15:
Hi,
Thank you for helping.
The problem is solved by updating CUDA to be version 12.2.

closed 2024-03-15 by maytusp

0mdc, 2024-03-15:
This is really helpful! Thanks for letting us know.

cuijiaxun, 2024-05-01:
@maytusp Have you tried any 11.{7, 8} cuda version?