facebookresearch/habitat-sim #2323: How to obtain region bounding boxes in HM3D?

frank-gallagher, 2024-02-21:
## Habitat-Sim version
v0.3.0 conda package & built from source main branch

## ❓ Questions and Help
 How do I obtain the region bounding boxes shown in @aclegg3  [Region Query API #2317](https://github.com/facebookresearch/habitat-sim/pull/2317)? Is this accessible through the python API?  The `SemanticRegion.volume_edges` property is an empty list.

Any help would be much appreciated.

aclegg3, 2024-02-21:
Hey @frank-gallagher,

Region bounding boxes are not presently annotated for HM3D because regions are annotated per-object rather than global geometry.

If you need these, you can compute them yourself from the objects in each region or manually annotate them and add the config files (see the test assets added with [this PR](https://github.com/facebookresearch/habitat-sim/pull/2299) for an example).

frank-gallagher, 2024-02-21:
Hi @aclegg3, I appreciate your quick response. Thanks for clearing that up!

Related to this question I'm having trouble exporting a CSV file with bounding box info from the viewer as you mention [here](https://github.com/matterport/habitat-matterport-3dresearch/issues/20#issuecomment-1300917988). 

If I launch the viewer with `habitat-viewer /mnt/data/habitat/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.glb` and then press the `J` key I get this error `Unable to build semantic CC objects since no semantic mesh exists`. However, a CSV file is still generated with bounding box coordinates but no semantic category information (I've included a snippet of the resulting file below). On the other hand if I launch the viewer with the semantic annotations using `habitat-viewer --dataset /mnt/data/habitat/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json TEEsavR23oF`, the `J` key binding does nothing and no CSV file is produced at all.

Firstly, what am I missing regarding how this CSV export is supposed to work?
Secondly, am I right in thinking this CSV only contains object level information or does it include region level bounding boxes also?

```
Obj IDX, Object ID, Color RGB, # Verts in partition, BBOX Center XYZ, BBOX Dims XYZ, BBOX Vol
0,Unknown_0,0 0 0,84,-4.58087 2.02348 -4.89537, 0.15872 0.158504 0.140015,0.00352247
0,Unknown_0,0 0 0,194,-4.24955 1.58655 -4.733, 0.0802779 0.594599 0.10411,0.00496951
0,Unknown_0,0 0 0,3,-6.04151 -5.10475e-05 -4.59289, 0.120018 0.001955 5.72205e-06,1.34259e-09
0,Unknown_0,0 0 0,3,-5.98281 2.73714 -4.90295, 0.00463486 0.00377131 0.299967,5.24326e-06
0,Unknown_0,0 0 0,29,-5.84325 0.0841004 -4.9529, 0.19651 0.0978 0.200001,0.00384376
0,Unknown_0,0 0 0,305,-5.60432 0.390835 -3.9079, 0.474372 0.582252 0.35,0.0966714
0,Unknown_0,0 0 0,3,-5.36151 0.345203 -3.63635, 3.8147e-06 0.139998 0.00273609,1.46121e-09
0,Unknown_0,0 0 0,16,-5.46556 1.3752 -3.67292, 0.0288248 0.679999 0.040098,0.000785954
0,Unknown_0,0 0 0,97,-4.57089 2.02371 -4.71172, 0.156851 0.17701 0.139235,0.00386574
0,Unknown_0,0 0 0,3,-4.4991 0.245224 -4.17291, 0.00325108 0.100018 5.29289e-05,1.72107e-08
0,Unknown_0,0 0 0,3,-4.48651 1.94517 -4.85289, 0.00407314 0.099996 1.09673e-05,4.46694e-09
0,Unknown_0,0 0 0,3,-5.37151 -0.011998 -3.97904, 0.099998 0.014431 0.052309,7.54857e-05
0,Unknown_0,0 0 0,3,-6.00229 1.2152 -4.05168, 0.00456381 0.04 0.0025301,4.61875e-07
0,Unknown_0,0 0 0,3,-6.19151 0.00118852 -4.03067, 0.0200019 0.028026 0.00444508,2.49179e-06
0,Unknown_0,0 0 0,3,-5.28923 -0.00442798 -4.04291, 0.144564 0.003563 0.0200286,1.03164e-05
0,Unknown_0,0 0 0,55,-5.7315 2.0952 -3.81605, 0.500006 0.11999 0.353696,0.0212202
0,Unknown_0,0 0 0,3,-5.82156 -0.087167 -4.4729, 0.000138283 0.015665 0.0799842,1.73262e-07
0,Unknown_0,0 0 0,3,-5.39151 0.00702289 -5.14052, 0.02 0.016355 0.0152597,4.99146e-06
0,Unknown_0,0 0 0,10,-5.48911 0.765202 -3.67289, 0.0471787 0.500002 0.0400181,0.000944005
...
...
...
```

closed 2024-02-29 by aclegg3