facebookresearch/habitat-sim #2327: Unable to add non primitive objects in habitat v0.3.0

norujumping, 2024-02-23:
## ❓ Questions and Help

I have installed habitat-sim v0.3.0 from source and was trying to add interactive rigid objects following [tutorial](https://aihabitat.org/docs/habitat-sim/managed-rigid-object-tutorial.html)

I also downloaded the necessary essets so that 

However, I get error in below lines
```
    observations = []
    obj_templates_mgr.load_configs(
        str(os.path.join(data_path, "objects/example_objects/"))
    )
    # search for an object template by key sub-string
    cheezit_template_handle = obj_templates_mgr.get_template_handles(
        "data/objects/example_objects/cheezit"
    )[0]
```

The files are downloaded correctly because `load_configs` is running without error, and the loading of spheres works without error.

I believe this is the same problem addressed in https://github.com/facebookresearch/habitat-sim/issues/951

It was resolved uisng version v0.1.7, but I think the problem persists in v0.3.0

Do we require different ways for loading custom objects to the scene, other than the method written in the tutorials?

aclegg3, 2024-02-23:
Hey @norujumping,

Have you tried enumerating the object templates available to check that they were loaded correctly and to identify the correct handle?

```
print(obj_templates_mgr.get_template_handles(""))
```

Should tell you what was actually loaded. Hard to help without knowing a bit more (e.g. seeing some log output)

norujumping, 2024-02-23:
Hey,

When I enumerate the object templates, I can't find any objects added
[image]

While this is what I have in the `objects/example_objects` folder
[image]

aclegg3, 2024-02-23:
The main issue here is that these objects have outdated configs. Habitat after 0.2.0 will not support ".phys_properties.json" files. Instead you should use ".object_config.json". So the objects are not being loading by the `load_configs` call.

You can get updated example objects here: http://dl.fbaipublicfiles.com/habitat/objects_v0.2.zip
Or a full dataset of YCB objects here: https://huggingface.co/datasets/ai-habitat/ycb (the example cheezit box is from YCB originally).

norujumping, 2024-02-23:
Thank you so much, it works with the ".object_config.json"

closed 2024-02-23 by norujumping