facebookresearch/habitat-sim #2328: datasets_download script not downloading semantic data for hm3d

janblumenkamp, 2024-02-29:
## Habitat-Sim version
v0.3.0

## 🐛 Bug

The download script does not seem to download the semantic information for the train splits in the hm3d dataset.

## Steps to Reproduce

Steps to reproduce the behavior:
Run the download script as
```
python -m habitat_sim.utils.datasets_download --username xxx --password xxx --uids hm3d_full
```

The semantic data exists for minival:
```
> ls data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.*
data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb      data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.semantic.glb
data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.navmesh  data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.semantic.txt
data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.glb
```

And for val:
```
> ls data/scene_datasets/hm3d/val/00800-TEEsavR23oF/TEEsavR23oF.*
data/scene_datasets/hm3d/val/00800-TEEsavR23oF/TEEsavR23oF.basis.glb      data/scene_datasets/hm3d/val/00800-TEEsavR23oF/TEEsavR23oF.semantic.glb
data/scene_datasets/hm3d/val/00800-TEEsavR23oF/TEEsavR23oF.basis.navmesh  data/scene_datasets/hm3d/val/00800-TEEsavR23oF/TEEsavR23oF.semantic.txt
data/scene_datasets/hm3d/val/00800-TEEsavR23oF/TEEsavR23oF.glb
```

But apparently not for train:
```
> ls data/scene_datasets/hm3d/train/00000-kfPV7w3FaU5/kfPV7w3FaU5.*
data/scene_datasets/hm3d/train/00000-kfPV7w3FaU5/kfPV7w3FaU5.basis.glb      data/scene_datasets/hm3d/train/00000-kfPV7w3FaU5/kfPV7w3FaU5.glb
data/scene_datasets/hm3d/train/00000-kfPV7w3FaU5/kfPV7w3FaU5.basis.navmesh
```

## Expected behavior

I expect the semantic data to exist for train.

## Additional context

I also notice that the file `data/scene_datasets/hm3d_semantic_v0.2/hm3d_basis.scene_dataset_config.json` lists scene instance descriptions `*.basis.scene_instance.json`, but this does not exist either, which results in a lot of warnings like

```
[10:19:42:933663]:[Warning]:[Metadata] AttributesManagerBase.h(398)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_semantic_v0.2/exa
mple/00861-GLAQ4DNUx5U/*.basis.scene_instance.json` so unable to load templates from that path.                                                                                                       
[10:19:42:933852]:[Warning]:[Metadata] AttributesManagerBase.h(398)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_semantic_v0.2/min
ival/00800-TEEsavR23oF/*.basis.scene_instance.json` so unable to load templates from that path.            
[...]                                                                                           
```

Furthermore, it seems like the test set is not downloaded at all:
```
[10:19:42:917474]:[Warning]:[Metadata] AttributesManagerBase.h(398)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d_semantic_v0.2/tes
t/00900-XRJ2muKkwAV/*.basis.glb` so unable to load templates from that path.                                                                                                                          
[10:19:42:917636]:[Warning]:[Metadata] AttributesManagerBase.h(398)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d_semantic_v0.2/tes
t/00901-aJoC8Qw6xQ5/*.basis.glb` so unable to load templates from that path.                                                                                                                          
```

## System Info
```
ENVIRONMENT INFO:
Platform: Linux-5.15.0-60-generic-x86_64-with-glibc2.35
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.35
Mac version: 
Python version: 3.9.18
Architecture: 64bit ELF
Win version:    
System OS: Linux
Release: 5.15.0-60-generic
Version: #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023
Operational System: linux
GCC version: b'gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0\n'
CMAKE version: b'cmake version 3.22.1\n'
NVIDIA-SMI: b'Thu Feb 29 10:53:46 2024       \n+-----------------------------------------------------------------------------+\n| NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0     |\n|-------------------------------+----------------------+----------------------+\n| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |\n| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |\n|                               |                      |               MIG M. |\n|===============================+======================+======================|\n|   0  NVIDIA A100-PCI...  On   | 00000000:21:00.0 Off |                    0 |\n| N/A   58C    P0    74W / 250W |  25641MiB / 40960MiB |     55%      Default |\n|                               |                      |             Disabled |\n+-------------------------------+----------------------+----------------------+\n|   1  NVIDIA A100-PCI...  On   | 00000000:81:00.0 Off |                    0 |\n| N/A   55C    P0    98W / 250W |  24479MiB / 40960MiB |     75%      Default |\n|                               |                      |             Disabled |\n+-------------------------------+----------------------+----------------------+\n'
Pip packages versions:
b''
Conda packages versions:
b''
```

aclegg3, 2024-02-29:
Hey @janblumenkamp,

Thanks for reaching out, a couple things here:
1. HM3D semantic annotations do not cover the full set of 1000 scenes. If you downloaded the semantic dataset, there should be config files limiting the dataset to those scenes which have annotations: `hm3d-train-semantic-configs-v0.2/hm3d_annotated_train_basis.scene_dataset_config.json`. The scene you referenced `kfPV7w3FaU5` is not in the list. Check one of these to validate that annotations were downloaded correctly.
2. `test` split assets are not available to public. We keep them in reserve such that future challenges can be run without biasing results by purposefully or accidentaly training or evaluating on them. Instead, do your own evaluation using the `val` set.

janblumenkamp, 2024-03-01:
Hey Alexander, thanks a lot for the quick response and clarifications!

Indeed, the semantic data is available for the scenes listed in that json file. I assume it's fine for me just to ignore these warnings then.

Ritabrata-Chakraborty, 2025-07-15:
@aclegg3  Would you know how to solve the issue mentioned below? The basis.scene_instance.json are not downloaded even when I use the link to the full val dataset. Any idea why? Could you maybe provide me with the link so I could confirm?

`[01:52:24:371487]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00832-qyAac8rV8Zk/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371510]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00835-q3zU7Yy5E5s/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371532]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00839-zt1RVoi7PcG/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371555]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00843-DYehNKdT76V/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371582]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00844-q5QZSEeHe5g/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371606]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00847-bCPU9suPUw9/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371627]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00848-ziup5kvtCCR/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371649]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00849-a8BtkwhxdRV/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371670]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00853-5cdEh9F2hJL/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371693]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00861-GLAQ4DNUx5U/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371715]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00862-LT9Jq6dN3Ea/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371738]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00869-MHPLjHsuG27/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371760]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00871-VBzV5z6i1WS/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371783]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00873-bxsVRursffK/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371805]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00876-mv2HUxq3B53/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371828]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00877-4ok3usBNeis/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371850]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00878-XB4GS9ShBRE/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371873]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00880-Nfvxx8J5NCo/*.basis.scene_instance.json` so unable to load templates from that path.
[... cut ...]
[01:52:24:371917]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00891-cvZr5TUy5C5/*.basis.scene_instance.json` so unable to load templates from that path.
[01:52:24:371939]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00894-HY1NcmCgn3n/*.basis.scene_instance.json` so unable to load templates from that path.
`