facebookresearch/habitat-sim #2329: Habitat-sim installation

QixuanSun, 2024-02-29:
Hello, I'm using the code from https://github.com/GT-RIPL/robo-vln, where a modified version of habitat-sim was used [habitat-sim @ d8f5e9b](https://github.com/zubair-irshad/habitat-sim/tree/d8f5e9b664ea28a1db88dbcb89dc9ae3e76d1410).

I was trying the following code to install habitat-sim, but errors occurred.
cd environments/habitat-sim
python setup.py install --headless --with-cuda

(habitat) [q84sun@gra-login1 habitat-sim]$ python setup.py install --headless --with-cuda
running install
running bdist_egg
running egg_info
creating habitat_sim.egg-info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sensor.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/geo.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/physics.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/gfx.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/attributes.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/registry.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/attributes_managers.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sim.py -> build/lib.linux-x86_64-3.6/habitat_sim
creating build/lib.linux-x86_64-3.6/habitat_sim/bindings
copying habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/bindings
creating build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/agent.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
creating build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/profiling_utils.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/viz_utils.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/collect_env.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/common.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/compare_profiles.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
creating build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
creating build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
creating build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/pyrobot_noisy_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/default_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/object_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
creating build/lib.linux-x86_64-3.6/habitat_sim/utils/data
copying habitat_sim/utils/data/data_extractor.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils/data
copying habitat_sim/utils/data/data_structures.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils/data
copying habitat_sim/utils/data/pose_extractor.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils/data
copying habitat_sim/utils/data/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils/data
creating build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/no_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/redwood_depth_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
[... 122 lines cut ...]
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 204, in run
    self.build_extension(ext)
  File "setup.py", line 273, in build_extension
    env=env,
  File "/home/q84sun/miniconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/project/6001557/q84sun/navigation/robo-vln/environments/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/project/6001557/q84sun/navigation/robo-vln/environments/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/q84sun/miniconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=ON']' returned non-zero exit status 1.

Could you please help me figure it out?

QixuanSun, 2024-03-01:
I've also tried to install with conda directly. 
E.g., conda install habitat-sim -c conda-forge -c aihabitat
or with a specific version, conda install habitat-sim=0.1.7 -c conda-forge -c aihabitat.
But both failed.
And here's the error message

Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - habitat-sim

Current channels:

  - https://conda.anaconda.org/robostack/osx-64
  - https://conda.anaconda.org/robostack/noarch
  - https://conda.anaconda.org/conda-forge/osx-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

shengyin1224, 2024-05-13:
If you use Macbook and have arm64 not X86-64, you can conda create -n env and then first input `conda config --env --set subdir osx-64 ` in the terminal.

vidyav2, 2024-06-10:
@realjoshqsun have you been able to resolve this issue? 
I'm stuck w/ the same issue, any help to resolve this is appreciated.

hjy0712, 2024-10-28:
i am stuck with the same issue

realjoshqsun, 2024-11-04:
@vidyav2 @hjy0712 
I'm using ubuntu22.04. And it seems like there's a discrepancy for the installed package.
You have to download the packages on Anaconda and install it manually.

closed 2024-11-04 by realjoshqsun