facebookresearch/habitat-sim #2330: Request for Access to Gibson Real and MP3D Real Datasets

kszpxxzmc, 2024-03-01:
I am currently working on a project related to your paper "Habitat-Matterport 3D Dataset (HM3D):
1000 Large-scale 3D Environments for Embodied AI" and I find your research highly valuable. However, the datasets mentioned in the paper, specifically the Gibson Real and MP3D Real datasets, require an application process for access. Given the urgency of my work, I was wondering if you could kindly assist in providing their image sets.

aclegg3, 2024-03-01:
Hey @kszpxxzmc, glad to hear you're interested in this work. 

I can't speak to the Gibson approval process, as that project is managed by another group. 

If you have submitted your access request for MP3D (or HM3D), you should be approved within 24 hours. If you are seeing any delay in that approval I can poke the approving organization to unblock you.

kszpxxzmc, 2024-03-01:
Thank you very much!

closed 2024-07-10 by aclegg3