facebookresearch/habitat-sim #2331: Build from source error with zlib

Yuxin916, 2024-03-04:
Hi! Thank you for your work!

I encountered problems when i am doing python setup.py install for habitat-sim after git checkout to tags/challenge-2022. The error log:

Install the project...
[ 24%] Built target corrade
[ 24%] Built target MagnumSceneGraph
[ 24%] Built target MagnumShaderTools
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/lib/libz.so.1.2.11
CMake Error at cmake_install.cmake:65 (file):
file INSTALL cannot copy file
"/home/xxxx/habitat-sim/build/deps/openexr/zlib-build/libz.so.1.2.11"
to "/usr/local/lib/libz.so.1.2.11": Permission denied.

It looks like a problem for zlib building. Could you give some suggestions on that?
Thank you!

Yuxin916, 2024-03-04:
Sorry i made a mistake. all good now

closed 2024-03-04 by Yuxin916

zachzhuu, 2024-11-17:
Hi Yuxin,
I encountered the same problem. Could you please share how to solve it?

NearianChen, 2025-05-18:
> Hi Yuxin, I encountered the same problem. Could you please share how to solve it?

hi,bro,  do u know how to resolve this problem now?

zachzhuu, 2025-05-19:
> > Hi Yuxin, I encountered the same problem. Could you please share how to solve it?
> 
> hi,bro, do u know how to resolve this problem now?

Hi @NearianChen, I haven’t figured out the solution yet either.

But recently, I was able to successfully build habitat-sim when using this [repo](https://github.com/Eku127/habitat-data-collector).
I’m not exactly sure why it worked, but it might be helpful to you too.

NearianChen, 2025-05-19:
> Hi [@NearianChen](https://github.com/NearianChen), I haven’t figured out the solution yet either.
>
> But recently, I was able to successfully build habitat-sim when using this [...]

hi, Thanks for reply. I solve the problem by this way:
1. use apt install zlib
2. clean build cache, and rebuild
commands here:
```
sudo apt install zlib1g zlib1g-dev

# cd /path/to/habitat-sim/
rm -rf build
python setup.py install --headless
```
It works for me, Thanks again!