facebookresearch/habitat-sim #2332: Problme in running ECCV_2020_Interactivity example

EftePA, 2024-03-05:
1. Following the tutorials I have installed habitat-sim in this order:
```
1. mkdir habitatDir
2. cd habitatDir 
3. git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
4. cd habitat-sim
5. conda create -n habitat python=3.9 cmake=3.14.0
6. conda activate habitat
7. conda install habitat-sim withbullet -c conda-forge -c aihabitat
```
To test it: 
```
1. Installed git-lfs: sudo apt install git-lfs
2. Inside habitatDir/habitat-sim downloaded (testing) 3D scenes: 
  python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path ./data/
3. Download example objects: python -m habitat_sim.utils.datasets_download --uids habitat_example_objects --data-path ./data/
4. Interactive testing (shows a saloon): 
  python examples/viewer.py --scene ./data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
5. Physical interactions:
python -m habitat_sim.utils.datasets_download --uids replica_cad_dataset
python -m habitat_sim.utils.datasets_download --uids replica_cad_baked_lighting
python examples/viewer.py --dataset data/replica_cad/replicaCAD.scene_dataset_config.json --scene apt_1  // Shows an apartment
```
2. Installing Habitat-lab:
```
//In the same level as habitat-sim
git clone --branch stable https://github.com/facebookresearch/habitat-lab.git
cd habitat-lab
pip install -e habitat-lab  # install habitat_lab
```
3. Install habitat-baselines.
```
pip install -e habitat-baselines 
```

4.  Run ECCV_2020_Interactivity.py locally via ```python3 ECCV_2020_Interactivity.py```
Then I am getting the error:
```
AssertionError: ESP_CHECK failed: Physics has been enabled in the SimulatorConfiguration but Habitat-Sim was not built with Bullet support enabled.

Either set cfg.enable_physics to False, or verify your Bullet installation (e.g recompile Habitat-Sim using the '--bullet' flag or choose a 'withbullet' conda build.)
```

As mentioned in step 1, habitat-sim is installed with Bullet. So I tried to set cfg.enable_physics to False, but didn't help.
Then I tried:

```
python setup.py --bullet build_ext --parallel 2 install
```
Now I am getting the error:
```
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 3 EGL devices in total
WindowlessContext: Unable to create windowless context
```
Any idea what I've made wrong and how I can fix it? Thank you.

isdelrey, 2024-03-08:
I have a similar error:
```

[15:38:24:975675]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_00.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975708]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_01.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975720]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_02.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975732]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_03.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975744]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_04.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975756]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_05.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975767]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_06.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975778]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_07.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975790]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_08.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975801]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_09.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975813]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_10.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975825]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_11.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975837]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_12.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975848]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_13.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975860]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_14.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975872]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_15.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975884]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_16.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975896]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_17.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975908]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_18.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975920]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_19.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[15:38:24:975932]:[Warning]:[Metadata] SceneDatasetAttributesManager.cpp(124)::loadAndValidateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_20.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 5 EGL devices in total
WindowlessContext: Unable to create windowless context

ERROR conda.cli.main_run:execute(124): `conda run python algorithm/muzero-cpp/main.py -n 1 -ep 10 -duration 12 -batch_size 10 -buffer_size 10` failed. (See above for error)
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_EXT_device_drm
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
[... cut ...]
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=3): EGL_EXT_device_drm
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=4): EGL_MESA_device_software
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
```

Does anyone know where the "WindowlessContext: Unable to create windowless context" issue might come from. I reinstalled the NVIDIA drivers and that did not work. 

Thanks in advance

EftePA, 2024-03-08:
I had not restarted my system after installing  CUDA Toolkit. Today, by accident, I restarted, and now it is working!
These are the steps I took to install cuda toolkit 12.4 on Ubuntu 22:

Base Installer:
```
wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64/cuda-ubuntu2204.pin
sudo mv cuda-ubuntu2204.pin /etc/apt/preferences.d/cuda-repository-pin-600
wget https://developer.download.nvidia.com/compute/cuda/12.4.0/local_installers/cuda-repo-ubuntu2204-12-4-local_12.4.0-550.54.14-1_amd64.deb
sudo dpkg -i cuda-repo-ubuntu2204-12-4-local_12.4.0-550.54.14-1_amd64.deb
sudo cp /var/cuda-repo-ubuntu2204-12-4-local/cuda-*-keyring.gpg /usr/share/keyrings/
sudo apt-get updatesudo apt-get -y install cuda-toolkit-12-4
```
Driver Installer
```
sudo apt-get install -y cuda-drivers
```

Hope helps others.

aclegg3, 2024-07-10:
Hey @EftePA ,

I'm glad to hear this is working! It can be a pain that CUDA drivers compile behind the scenes and may require a system restart to take affect.

I'm going to close this issue because the original problem is solved.

> now example.py inside habitat-sim is working (but not the example.py in habitat-api)!
It is worth noting that habitat-api is no longer a functioning repo. It was converted to habitat-lab.

closed 2024-07-10 by aclegg3