facebookresearch/habitat-sim #2335: How to Render Replica Dataset Scenes with habitat-viewer?

Liu-Jinxin, 2024-03-12:
## Issue: Problems Rendering Replica Dataset Scenes with habitat-viewer

### Description
I am trying to use Habitat Sim to visualize scenes from the Replica dataset but encountering rendering issues. Specifically, the visuals appear incorrect when executing the command for the Replica dataset, while rendering works as expected for the ReplicaCAD dataset.

### Commands and Results
- **Replica Dataset Issue Command:**
  ```bash
  habitat-viewer --enable-physics --dataset data/Replica-Dataset/data/replica.scene_dataset_config.json --room_1
 ```
[image]

aclegg3, 2024-03-12:
We have removed support for ptex shader which was handling Replica scenes specifically. I imagine this is related to that deprecation.
I'm a bit surprised to see they are loading with what looks like base metallic color and no texture, though.
@jturner65 any thoughts on this?

Liu-Jinxin, 2024-03-13:
> We have removed support for ptex shader which was handling Replica scenes specifically. I imagine this is related to that deprecation. I'm a bit surprised to see they are loading with what looks like base metallic color and no texture, though. @jturner65 any thoughts on this?

Thank you for the insight on the deprecation of ptex shader support, given this context, could you please inform me about the last version or branch of Habitat Sim that fully supported normal rendering for the Replica dataset with the ptex shaders? I'm looking to work with the Replica dataset.

Thank you for your assistance.

emilia-szymanska, 2024-05-29:
@Liu-Jinxin Hi! Did you perhaps manage to figure this out?

Liu-Jinxin, 2024-05-30:
> Thank you for the insight on the deprecation of ptex shader support, given this context, could you please inform me about the last version or branch of Habitat Sim that fully supported normal rendering for the Replica dataset with the ptex shaders? I'm looking to work with the Replica dataset.

No, I changed the version.