facebookresearch/habitat-sim #2343: How to view the navmesh in 3D?

frank-gallagher, 2024-03-22:
## ❓ Questions and Help

Rather than rendering topdown views of the navmesh as shown in the tutorials, how do I view the navmesh in 3D?

aclegg3, 2024-03-22:
Hey @frank-gallagher I have two options for you:
1. You can set the [Simulator.navmesh_visualization](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#navmesh_visualization) attribute to `True`, which adds a wireframe representation to all sensor images.
2. You can query the navmesh [vertices](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#build_navmesh_vertices) and [indices](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#build_navmesh_vertex_indices) (for the full navmesh or any individual island) and then use the [DebugLineRender](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#get_debug_line_render) to manually draw the mesh edges in any color you want. This would need to be re-drawn before each sensor is rendered (e.g. see the [viewer example](https://github.com/facebookresearch/habitat-sim/blob/5f2bc6a0bde1da77b334022ca28e8903952f4a24/examples/viewer.py#L309))

frank-gallagher, 2024-06-07:
Hi @aclegg3 ,

The _[build_navmesh_vertex_indices](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#build_navmesh_vertex_indices)_ function just returns a list of integers from 0 to the length of the list returned by _[build_navmesh_vertices](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#build_navmesh_vertices)_. Should it not provide the triangle index for each vertex or am I misunderstanding something?