facebookresearch/habitat-sim #2350: Rendering with ScanNet

xz6014, 2024-03-27:
Hi there,

Hope that you could help. 

I was trying to render a scannet scene with the COLOR SensorType, and I was getting the following error and warnings:

[16:14:25:206625]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset : Dataset : 'default' : Lighting Layout Attributes 'no_lights' specified in Scene Attributes but does not exist in dataset, so creating default.
[16:14:25:206832]:[Warning]:[Scene] SemanticScene.h(328)::checkFileExists : ::loadSemanticSceneDescriptor: File databases19/ScanNet/scans/scene0000_00/scene0000_00_vh_clean_2.scndoes not exist.  Aborting load.                         ult.                                
[16:14:25:207037]:[Warning]:[Scene] SemanticScene.cpp(121)::loadSemanticSceneDescriptor : SSD File Naming Issue! Neither SemanticAttributes-provided name : `databases19/ScanNet/scans/scene0000_00/scene0000_00_vh_clean_2.scn` nor const                                       ructed filename : `databases19/ScanNet/scans/scene0000_00/info_semantic.json` exist on disk.
[16:14:25:207088]:[Error]:[Scene] SemanticScene.cpp(137)::loadSemanticSceneDescriptor : SSD Load Failure! File with SemanticAttributes-provided name `databases19/ScanNet/scans/scene0000_00/scene0000_00_vh_clean_2.scn` exists but failed to load.
[16:14:25:364030]:[Warning]:[Sim] Simulator.cpp(594)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0 

Thanks in advance for your support!

0mdc, 2024-06-04:
@aclegg3 , @jturner65 : Would you have some context on this?

aclegg3, 2024-06-04:
Our scannet dataset use instructions can be found here: https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#scannet (not sure if things have changed since this was written).

From what I see here, I would suggest you check the "scene" path:
`"scene": "scene0015_00_vh_clean.glb",`
This should actually be the relative path to the asset (from the working directory) unless you have a `scene_dataset_config.json` referencing the directory or paths to the assets within.

See https://aihabitat.org/datasets/replica_cad/ for an example of setting up SceneDataset configs.

outof-memory, 2024-08-17:
> Our scannet dataset use instructions can be found here: https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#scannet (not sure if things have changed since this was written).
>
> From [...]

“scene”path already set “data/scene_datasets/habitat-test-scenes/skokloster-castle.glb'
but 
1）the error message still exists
`[08:26:07:801824]:[Error]:[Scene] SemanticScene.cpp(137)::loadSemanticSceneDescriptor : SSD Load Failure! File with SemanticAttributes-provided name `data/scene_datasets/habitat-test-scenes/apartment_1.scn` exists but failed to load.`
2）rgb sensor output images with black background，only sevral objects of apsule3DSolid_hemiRings_4_cylRings