facebookresearch/habitat-sim #2351: distance_to_closest_obstacle() return wrong distance while getting close to stage

Fanxing-LI, 2024-03-28:
## Habitat-Sim version
v0.3.1

I have created a self-defined drone dynamics system using habitatsim as render. And I obtain the collision status by comparing the pathfinder.distance_to_closest_obstacle() to the radius of my ball-shape agent. 

I have confirmed that the "collision_assert" (the same as render_assert) is included in configs/stages/**.stage_config.json, "is_collidable" is set as true. Besides, I've double check these settings after creating the Simulator.

the issue is that, the distance returned from pathfinder.distance_to_closest_obstacle(_point_, max_search_radius) is correct while the _point_ is getting close to walls and objects, but wrong while getting close to ground/floor. It seems that the agents do not consider the ground as obstacles, assuming that robot always stands on the floor.

If I want to free the point in the room and make it aware of the closest distance in any direction (including ground), how could I do.
I appreciate it very much if you can offer any help.

Thanks!

aclegg3, 2024-04-05:
Hey @Fanxing-LI 

Interesting application. A few points of clarification about how the system works may help:
1. The navmesh (PathFinder) is an abstraction of the navigable area in a scene representing the agent as a cylinder and computing a mesh based on a voxelization of the scene geometry. 
2. The `distance_to_closest_obstacle` function gives distance to the nearest mesh edge from the snapped point. 
3. Pathfinder does not use the collision proxy for the asset, it only uses the NavMeshSettings which by default is computed from the agent's height and radius config.
4. The navmesh will not be a great representation if you want to the drone to be able to fly/hover over obstacles in addition to avoiding them.

It sounds like you have a clever way of using the navmesh for a drone to avoid walls and large obstacles, but you want vertical distance (e.g. to floor and maybe ceiling) in addition. One way to accomplish this is to setup some[ raycasts ](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#cast_ray) as sensors. For example, [cast a ray](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#cast_ray) from the center of your drone down the Y-axis and use the first hit distance (which isn't your drone). Similarly, you could setup rays in other directions for additional sensors.

Another option (if the world is mostly static, like scan scenes) would be to pre-compute a signed distance field in the space and then query from the cache for your drone's position.

Fanxing-LI, 2024-04-18:
Thanks for your reply. Besides obove-mentioned issue, closest_obstacle_surface_point() returns the obstacles' position projected on the ground, which is also not right for collision detection in free space. 
So we decide to modify the source code and try to add a self-defined function to compute the vector (to), distance (to) ,position (of) closest obstacle surface point, without constraints. For this task, the original mesh map or point cloud map is necessary. If possible, could you tell me which part/ variable is the original map resourses? Thanks