facebookresearch/habitat-sim #2370: where is the “datatool”?

zhai-create, 2024-04-25:
I want to generate semantic information for gibson (. py file) through the gen_gibson_semantics.sh file. During the generation process, I need a "datatool" file, but I did not find datatool in all the code. Where is this datatool? If it is obtained through compilation, how should I compile it?

## ❓ Questions and Help

aclegg3, 2024-04-25:
Hey @zhai-create,
Yes, you will need to add a build flag so the datatool is compiled.
If building with setup.py you can add `--build-datatool`

zhai-create, 2024-04-25:
@aclegg3 Thank you. However, I have a new quesrtion in the process of generating the semantic datasets. I have created a new question box in the Issue of the current project, titled "Error in generation of semantic datasets".