facebookresearch/habitat-sim #2372: Error in generation of semantic datasets

zhai-create, 2024-04-25:
When I generated semantic data for gibson using the gen_gibson_semantics.sh file, I found that there were some scenarios where the following errors occurred:

`terminate called after throwing an instance of 'std::length_error'
what():  cannot create std::vector larger than max_size()
Aborted (core dumped)`

How should I solve this problem? Do I need to modify the `Datatool.cpp` file again?（I have generated `.ids` files, `.obj` files, and `datatool` for each scene）

## ❓ Questions and Help

iyeon915, 2024-05-27:
I got a similar issue. Did you solved this issue?
I obtained build foler using "python setup.py build --build-datatool" at here (https://github.com/facebookresearch/habitat-sim/issues/1948).
Then the datatool file in build foler was build/utils/datatool/Datatool not build/utils/datatool/datatool. (Capital D)
So, I modified gen_gibson_semantic.sh file, then run using "tools/gen_gibson_semantics.sh /path/to/3DSceneGraph_medium/automated_graph /path/to/GibsonDataset /path/to/output
".

There are 3 folders in scene datasets folder, 
1) 3DSceneGraph_medium (includes npz)
2) gibson (includes scene_dataset_config.json and scene folder. Each scene folder has obj files)
3) out folder (.ids and .scn files were generated. but .ply is not)

Finally the error is below.
(habitat) user@server90:~/habitat/habitat-sim$ tools/gen_gibson_semantics.sh data/scene_datasets/3DSceneGraph_medium/automated_graph/ data/scene_datasets/gibson/ data/scene_datasets/out/
Adairsville
wrote 2528538 bytes
wrote 33569 bytes
ESP_CHECK failed: esp::logging::LoggingContext: No current logging context.
tools/gen_gibson_semantics.sh: line 17: 22515 Aborted                 (core dumped) "${TOOLS_DIR}"/../build/utils/datatool/datatool create_gibson_semantic_mesh "${OBJ_PATH}"/"${scene}"/mesh.obj
"${OUT_PATH}"/"${scene}".ids "${OUT_PATH}"/"${scene}"_semantic.ply

iranroman, 2024-06-13:
Also having the exact same issue. Was any of you able to solve it?

yuanxuanS, 2024-09-23:
I followed this steps: https://github.com/ybgdgh/L3MVN/issues/22#issuecomment-2057438619 and succeed finally.