facebookresearch/habitat-sim #2374: the lighting of "dataset mode" and "scene mode" are different, how to change it

Jay-Vim-Lv, 2024-04-26:
## Habitat-Sim version
v0.3.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/
yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
yes
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I am new learner, when I try to run the examples/viewer.py, i found this situation.

i used the hssd-hab dataset and the code like this
`
--dataset data/.../hssd-hab.scene_dataset_config.json
--scene data/.../103997919_171031233.scene_instance.json
`
the scene is like this:
[image]

and it looks different from the picture in the paper or glb in the dataset. (As known, hssd-hab is the compressed version of hssd)

so i download the hssd dataset, and run it like this
`
--scene data/.../103997919_171031233.glb
`
the scene is like this:
[image]

so i choose a component, hssd-hab/stages/103997919_171031233.glb
`
--scene data/.../103997919_171031233.glb
`
the scene is like this：
[image]
I don't know how to describe it exactly, it seems like the lighting or color is different. how does it happen? and what should i do to set it.

waiting for the help.

aclegg3, 2024-04-26:
Hey @Jay-Vim-Lv,

That one scene you are using does have a missing kitchen element (top image), so that is a known issue.

You may be looking for the shadows provided by Horizon Based Ambient Occlusion(HBAO). You can toggle this with `--hbao` in the python viewer.

For example, the same scene with HBAO:
[image]

Jay-Vim-Lv, 2024-04-27:
Thank you for the reply @aclegg3 !
 I added the "--hbao" like this:
 `
--dataset data/.../hssd-hab.scene_dataset_config.json 

--scene data/.../103997919_171031233.scene_instance.json

--hbao
`
I'm sure `sim_settings["enable_hbao"]` is `True`
but like the scene still looks different like yours:
[image]

aha,  it seems the same as  the top image, what is this about?

Jay-Vim-Lv, 2024-04-27:
Thank you for the reply @aclegg3 !
I added the "--hbao" like this:

```
--dataset data/.../hssd-hab.scene_dataset_config.json

--scene data/.../103997919_171031233.scene_instance.json

--hbao
```

I'm sure `sim_settings["enable_hbao"]` is `True` at this time.
but like the scene still looks different like yours:
[image]

aha, it seems the same as the top image, what is this about?