facebookresearch/habitat-sim #2380: ReplicaCAD not loading articulated counters outside the viewer

jackkolb, 2024-04-29:
## Habitat-Sim version
v0.3.1

## ❓ Questions and Help

Hi! I am trying to render an apartment from the ReplicaCAD dataset, however am unable to load the articulated counters. I installed habitat-sim with bullet physics, however am receiving errors that suggest bullet physics are not enabled.

For example:

[image]

The counters are visible and interactable in the interactive viewer example, however when using my own config (based on the Shortest Path Follower example) I get these errors:

```
[18:42:48:737442]:[Warning]:[Metadata] MetadataMediator.cpp(587)::getFilePathForHandle : <getNavmeshPathByHandle> : Unable to find file path for empty_stage_navmesh , so returning empty string.
[18:42:48:948451]:[Warning]:[Sim] Simulator.cpp(594)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0
[18:42:50:542760]:[Error]:[Physics] PhysicsManager.h(1074)::addArticulatedObjectInternal : Not implemented in base PhysicsManager. Install with --bullet to use this feature.
[18:42:50:542785]:[Error]:[Physics] PhysicsManager.cpp(549)::addArticulatedObjectAndSaveAttributes : Articulated Object create failed for ArticulatedObjectAttributes 'data/scene_datasets/replica_cad/urdf/fridge/fridge.ao_config.json', whose handle contains 'fridge' as specified in articulated object instance attributes, so addArticulatedObjectInstance aborted.
[18:42:50:542824]:[Error]:[Physics] PhysicsManager.h(1074)::addArticulatedObjectInternal : Not implemented in base PhysicsManager. Install with --bullet to use this feature.
[18:42:50:542833]:[Error]:[Physics] PhysicsManager.cpp(549)::addArticulatedObjectAndSaveAttributes : Articulated Object create failed for ArticulatedObjectAttributes 'data/scene_datasets/replica_cad/urdf/kitchen_counter/kitchen_counter.ao_config.json', whose handle contains 'kitchen_counter' as specified in articulated object instance attributes, so addArticulatedObjectInstance aborted.
[18:42:50:542859]:[Error]:[Physics] PhysicsManager.h(1074)::addArticulatedObjectInternal : Not implemented in base PhysicsManager. Install with --bullet to use this feature.
[18:42:50:542867]:[Error]:[Physics] PhysicsManager.cpp(549)::addArticulatedObjectAndSaveAttributes : Articulated Object create failed for ArticulatedObjectAttributes 'data/scene_datasets/replica_cad/urdf/kitchen_cupboards/kitchenCupboard_01.ao_config.json', whose handle contains 'kitchenCupboard_01' as specified in articulated object instance attributes, so addArticulatedObjectInstance aborted.
[18:42:50:542892]:[Error]:[Physics] PhysicsManager.h(1074)::addArticulatedObjectInternal : Not implemented in base PhysicsManager. Install with --bullet to use this feature.
[18:42:50:542900]:[Error]:[Physics] PhysicsManager.cpp(549)::addArticulatedObjectAndSaveAttributes : Articulated Object create failed for ArticulatedObjectAttributes 'data/scene_datasets/replica_cad/urdf/chest_of_drawers/chestOfDrawers_01.ao_config.json', whose handle contains 'chestOfDrawers_01' as specified in articulated object instance attributes, so addArticulatedObjectInstance aborted.
[18:42:50:542926]:[Error]:[Physics] PhysicsManager.h(1074)::addArticulatedObjectInternal : Not implemented in base PhysicsManager. Install with --bullet to use this feature.
[18:42:50:542937]:[Error]:[Physics] PhysicsManager.cpp(549)::addArticulatedObjectAndSaveAttributes : Articulated Object create failed for ArticulatedObjectAttributes 'data/scene_datasets/replica_cad/urdf/cabinet/cabinet.ao_config.json', whose handle contains 'cabinet' as specified in articulated object instance attributes, so addArticulatedObjectInstance aborted.
[18:42:50:542958]:[Error]:[Physics] PhysicsManager.h(1074)::addArticulatedObjectInternal : Not implemented in base PhysicsManager. Install with --bullet to use this feature.
[18:42:50:542966]:[Error]:[Physics] PhysicsManager.cpp(549)::addArticulatedObjectAndSaveAttributes : Articulated Object create failed for ArticulatedObjectAttributes 'data/scene_datasets/replica_cad/urdf/doors/door2.urdf', whose handle contains 'door2' as specified in articulated object instance attributes, so addArticulatedObjectInstance aborted.
```

I installed habitat-sim from source using --bullet, and also tried on a new environment using the conda package (`withbullet`), following the README's installation directions. Same results.

What could be wrong? While the errors suggest an installation issue, the counters showing on the interactive viewer indicates a configuration issue.

My config.yaml is:

```
# @package _global_

defaults:
  - /habitat: habitat_config_base
  - /habitat/task: pointnav
  - /habitat/simulator/sensor_setups@habitat.simulator.agents.main_agent: rgbd_agent
  - /habitat/dataset: dataset_config_schema
  - _self_

habitat:  
  dataset:
    type: PointNav-v1
    data_path: "hab_1_episodes.json.gz"
    scenes_dir: data/scene_datasets/replica_cad
  environment:
    max_episode_steps: 100
  simulator:
    additional_object_paths:
        - "data/objects/ycb/configs/"
    agents:
      main_agent:
        sim_sensors:
          rgb_sensor:
            width: 256
            height: 256
          depth_sensor:
            width: 256
            height: 256
```

And my episode JSON is:

```
{
    "episodes": 
    [
        {
            "scene_id": "configs/scenes/apt_2.scene_instance.json",
            "scene_dataset_config": "data/scene_datasets/replica_cad/replicaCAD.scene_dataset_config.json",
            "episode_id": "0", 
            "start_position": [1.0, 0.2047852873802185, 1.0], 
            "start_rotation": [0, 0.4536385088584658, 0, 0.8911857849408661],
            "goals": [
                {
                    "position": [2.2896811962127686, 0.11950381100177765, 2.97636604309082]
                }
            ]
        }
    ]
}
```

aclegg3, 2024-04-30:
Hey @jackkolb 

Note this is an issue with habitat-lab configuration, not habitat-sim.

However, simply fix I think. It looks like you are using a default pointnav config. [By default, pointnav does not enable physics](https://github.com/facebookresearch/habitat-lab/blob/c3b4eede94d2078cdc75cd2dcf374cec80aa34ac/habitat-lab/habitat/config/default_structured_configs.py#L1714) since scanned static scenes have no need for dynamic simulation or collision detection.

Set that `enable_physics` flag to `True` and it should fix your problem.

jackkolb, 2024-04-30:
Thank you! That fixed it (and hopefully this helps someone down the line).

closed 2024-04-30 by jackkolb