facebookresearch/habitat-sim #2383: tutorial for using HM3D

sky-story, 2024-05-06:
Hello, I'm wondering if there's a comprehensive tutorial available to assist me in utilizing the Habitat Matterport 3D dataset within the Habitat framework? 
Could you also advise me on which sections of the codebase I should focus on for this purpose?
 I found the information provided in habitat-sim/DATASETS.md to be insufficient for effectively utilizing HM3D and other datasets.（sad）

aclegg3, 2024-05-06:
Hey @sky-story we'll need more information to help out here.

The [documentation for HM3D with Habitat](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#habitat-matterport-3d-research-dataset-hm3d) describes how to acquire the data locally and then how to configure the Simulator to load the scenes. 

Beyond that, your code is application specific. If you know what you want to do we may have more pointers for you. If you are new to AI Habitat, I suggest you take a look at some of our resources pages and tutorials [here](https://aihabitat.org/docs/habitat-sim/).

sky-story, 2024-05-07:
Thank you for your prompt response!
 It appears that the documentation for HM3D with Habitat has not been updated to guide users on utilizing HM3D via Habitat Lab. For instance, the documentation states that 'SIMULATOR.AGENT_0.SENSORS = ["RGB_SENSOR", "SEMANTIC_SENSOR"]', but this specific configuration is not present in the latest habitat-lab .yaml files！
Could you please provide a complete example code demonstrating how to utilize HM3D and visualize the observations (RGBD + semantic) in Habitat Lab? Or could you guide me on which parts of Habitat I should learn to achieve this？

sky-story, 2024-05-07:
Hey @aclegg3 
Thank you for your prompt response!
It appears that the documentation for HM3D with Habitat has not been updated to guide users on utilizing HM3D via Habitat Lab. For instance, the documentation states that 'SIMULATOR.AGENT_0.SENSORS = ["RGB_SENSOR", "SEMANTIC_SENSOR"]', but this specific configuration is not present in the latest habitat-lab .yaml files！
Could you please provide a complete example code demonstrating how to utilize HM3D and visualize the observations (RGBD + semantic) in Habitat Lab? Or could you guide me on which parts of Habitat I should learn to achieve this？