facebookresearch/habitat-sim #2393: Using the Python viewer and the Replica dataset with the sorted faces

emilia-szymanska, 2024-05-10:
Hello!
My goal is to find the simplest solution possible to just manually walk around a Replica apartment and take "screenshots" of what I see. For that purpose, I modified the `examples/viewer.py` so that when I press a corresponding button, I save the frame (either depth, semantic or RGB). However, I would like to have the nice, high quality scenes, not the ones made from scans. I browsed through the issues (like #709 or #1256), I have the sorted faces downloaded and placed in the habitat directory. I tried pointing at both `mesh.ply` and `habitat/mesh_semantic.ply`, but apparently it is not the solution, since it is not working. I do not have CUDA, so I tried to stick with the viewer, which works without it. Can I somehow modify the viewer to achieve my goal?

EverNorif, 2024-10-02:
hello, have you solved this problem?

EverNorif, 2024-10-02:
@emilia-szymanska  Thanks for your quick reply. It is indeed difficult to find detailed and up-to-date documentation :(

aclegg3, 2024-10-02:
Hey folks, sorry to hear about your trouble here and apologies for not addressing this sooner.

We did deprecate ptex shader support last year: https://github.com/facebookresearch/habitat-sim/pull/2137 so the source shaders expected to get maximum quality from the scan metadata won't be available unless you roll back to that version of the code.

Did you try loading the dataset from the config file as suggested [here](https://github.com/facebookresearch/Replica-Dataset?tab=readme-ov-file#replica-and-ai-habitat)?

I have not run Replica locally in awhile (small dataset for a large memory footprint) but I think it should still work.