facebookresearch/habitat-sim #2405: Python viewer runs perfectly, but other scripts end with Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total WindowlessContext: Unable to create windowless context

emilia-szymanska, 2024-06-07:
At this point I truly do not know what I am doing wrong. I do not have a GPU, but a python example viewer runs perfectly. However, if I try to do anything in another script, it fails with `Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
WindowlessContext: Unable to create windowless context`. I truly do not understand how a fancier script can run without problems, and the following piece breaks:

```python
from typing import Any, Dict
import habitat_sim
from habitat_sim.utils.settings import default_sim_settings, make_cfg

sim_settings: Dict[str, Any] = default_sim_settings
sim_settings["scene"] = "./data/test_assets/scenes/simple_room.glb"
sim_settings["scene_dataset_config_file"] = "default"
sim_settings["enable_physics"] = False
sim_settings["num_environments"] = 1
sim_settings["window_width"] = 800
sim_settings["window_height"] = 600
sim_settings["default_agent_navmesh"] = False
sim_settings["color_sensor"] = True
sim_settings["depth_sensor"] = True
sim_settings["semantic_sensor"] = True

cfg = make_cfg(sim_settings)
sim = habitat_sim.Simulator(cfg)
```
Please help :(

0mdc, 2024-06-10:
That's pretty strange. Your script works on my end.

>I do not have a GPU, but a python example viewer runs perfectly.

Which example viewer did you run?

emilia-szymanska, 2024-06-11:
@0mdc I ran `examples/viewer.py` with exactly this command:  `python examples/viewer.py --scene path/to/.ply`.

emilia-szymanska, 2024-06-20:
@0mdc why would you close this issue? It is very far from being solved.

0mdc, 2024-06-20:
My mistake, I closed the wrong one :)

Eku127, 2024-11-15:
Any solution yet? Meet the same problem with docker.

yi-fan-li, 2025-04-20:
I've having the same issue, running the interactive viewer from the example works, but running asset viewer tutorial or calling the simulator doesn't

Eku127, 2025-04-21:
> I've having the same issue, running the interactive viewer from the example works, but running asset viewer tutorial or calling the simulator doesn't

For me I solved by using the GLX instead of the OGL = =