facebookresearch/habitat-sim #2406: Error Downloading HM3D Dataset: tarfile.ReadError

MrCapricornLiu, 2024-06-08:
## Habitat-Sim Version
v0.3.1

## 🐛 Bug Report

I encountered an issue while following the instructions provided [here](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#matterport3d-mp3d-dataset) to download the HM3D dataset using the command:
```
python -m habitat_sim.utils.datasets_download --username <api-token-id> --password <api-token-secret> --uids hm3d_minival_v0.2
```
However, there are some problems during the download process. Please see the detailed error in the screenshot below:
```
  File "/root/miniconda3/envs/habitat_latest/lib/python3.9/tarfile.py", line 1797, in open
    raise ReadError("file could not be opened successfully")
tarfile.ReadError: file could not be opened successfully
```
[image]

## Steps to Reproduce

To reproduce the behavior, run the following command (the api-token is got from [here](https://my.matterport.com/settings/account/devtools): Developer Tools -> API Token Management):
```
python -m habitat_sim.utils.datasets_download --username 038b1d8ac662cb50 --password f3f88b6f885c882625ae34cab9d58ba1 --uids hm3d_minival_v0.2 --data-path /root/autodl-tmp/project/download
```

## System Info

- **OS:** Ubuntu 20.04
- **PyTorch Installation Method:** conda
- **Build Command (if compiling from source):**
- **Python Version:** 3.9
- **CUDA Version:** 11.0

ZiruiLi-133, 2024-06-26:
I have also met the same issue, this might happen because we did not have access to the dataset so basically nothing is downloaded. After I follow the instructions mentioned at https://github.com/matterport/habitat-matterport-3dresearch?tab=readme-ov-file and applied for the access, the problem was solved.

aclegg3, 2024-07-10:
Hey @MrCapricornLiu,

It is true that you will need access first. @ZiruiLi-133 I'm glad to hear this resolved the problem for you.

This can also result from temporary outages of the server resulting in corrupted or empty packages. Please ensure you have access and try again. If that doesn't resolve the issue we can dig in more.

closed 2024-08-14 by MrCapricornLiu