facebookresearch/habitat-sim #2413: Import Error

WYXG233, 2024-06-21:
## Habitat-Sim version
RLRAudioPropagationUpdate & habitat_sim-0.3.1

## 🐛 Bug

when I try to import habitat_sim.sim, it will output a TypeError. Below is the Traceback.

## Steps to Reproduce

1. python
2. import habitat_sim.sim

```
File "miniconda/envs/ss/lib/python3.9/site-packages/habitat_sim-0.3.1-py3.9-linux-x86_64.egg/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim import (
  File "miniconda/envs/ss/lib/python3.9/site-packages/habitat_sim-0.3.1-py3.9-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *  # noqa: F403
  File "miniconda/envs/ss/lib/python3.9/site-packages/habitat_sim-0.3.1-py3.9-linux-x86_64.egg/habitat_sim/agent/agent.py", line 12, in <module>
    import quaternion as qt
  File "miniconda/envs/ss/lib/python3.9/site-packages/quaternion/__init__.py", line 34, in <module>
    from .calculus import (
  File "miniconda/envs/ss/lib/python3.9/site-packages/quaternion/calculus.py", line 462, in <module>
    from scipy.interpolate import InterpolatedUnivariateSpline
  File "miniconda/envs/ss/lib/python3.9/site-packages/scipy/interpolate/__init__.py", line 167, in <module>
    from ._interpolate import *
  File "miniconda/envs/ss/lib/python3.9/site-packages/scipy/interpolate/_interpolate.py", line 14, in <module>
    from . import _fitpack_py
  File "miniconda/envs/ss/lib/python3.9/site-packages/scipy/interpolate/_fitpack_py.py", line 8, in <module>
    from ._fitpack_impl import bisplrep, bisplev, dblint  # noqa: F401
  File "miniconda/envs/ss/lib/python3.9/site-packages/scipy/interpolate/_fitpack_impl.py", line 103, in <module>
    'iwrk': array([], dfitpack_int), 'u': array([], float),
TypeError
```

## Expected behavior

import correctly

## System Info

Platform: Linux-6.5.0-35-generic-x86_64-with-glibc2.35
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.35
Mac version: 
Python version: 3.9.19
Architecture: 64bit ELF
Win version:    
System OS: Linux
Release: 6.5.0-35-generic
Version: #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2
Operational System: linux
GCC version: b'gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0\n'
CMAKE version: b'cmake version 3.14.0\n'
NVIDIA-SMI: b"NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.\n"
Pip packages versions:
b'imageio==2.34.1\nimageio-ffmpeg==0.5.1\nmatplotlib==3.8.4\nnumba==0.59.1\nnumpy==1.26.4\nnumpy-quaternion==2023.0.3\ntqdm==4.66.4\n'
Conda packages versions:
b'attrs                     23.2.0             pyh71513ae_0    conda-forge  19.1.0\ngitpython                 3.1.43             pyhd8ed1ab_0    conda-forge\nimageio                   2.34.1             pyh4b66e23_0    conda-forge\nimageio-ffmpeg            0.5.1              pyhd8ed1ab_0    conda-forge\nmatplotlib                3.8.4            py39hf3d152e_2    conda-forge\nnumba                     0.59.1           py39h615d6bd_0    conda-forge\nnumpy                     2.0.0                    pypi_0    pypi\npillow                    8.3.2            py39ha612740_0    conda-forge\nscipy                     1.13.1           py39haf93ffa_0    conda-forge  1.3.0\ntqdm                      4.66.4             pyhd8ed1ab_0    conda-forge\n'

ziyuwan, 2024-06-25:
@WYXG233 have you solved it now? I've also encountered this problem.

aclegg3, 2024-06-25:
This looks like a likely dependency version incompatibility between numpy, numpy-quaternion, and numba.

WYXG233, 2024-06-26:
@ziyuwan Yeah, I haven't found out how to fix it. But it works correctly in another environment. Maybe you can try the following dependency version: 
```
numpy-1.23.5 (instead of the above numpy 2.0.0)
```
btw, hope you can update this incompatibility issue with a higher version of numpy.

ziyuwan, 2024-06-26:
@WYXG233 Thanks a lot for your reply! 
I've also found that my `numpy` version was wrong. After installing `numpy==1.23.0`, it works now.

aclegg3, 2024-06-26:
For what its worth, my current build is working with:
```
numba                     0.60.0rc1                pypi_0    pypi
numpy                     1.26.4                   pypi_0    pypi
numpy-quaternion          2023.0.3                 pypi_0    pypi
```