facebookresearch/habitat-sim #2414: Unable to Navigate to Random Navigable Point

yrraadi-io, 2024-06-21:
I'm working with the `home-robot` repository currently and as part of the project I'm experimenting with navigating to random points in my current environment using `find_path` method in `GreedyGeodesicFollower`.

1. Initializing `ShortestPathFollower`:
`self.spf = ShortestPathFollower(sim, 0.25, False)`
Where I'm passing `sim=self._env.habitat_env.env._sim`

2. Getting a random navigable point:
```
self.spf._build_follower()
goal_pos = self.spf._follower.pathfinder.get_random_navigable_point(10)
```
3. Finding path to the point (In a seperate method by passing `goal_pos`)
```
self.spf._build_follower()
next_action = self.spf._follower.find_path(goal_pos=goal_pos)
```

However, this results in the following error:
```
raise errors.GreedyFollowerError()
habitat_sim.errors.GreedyFollowerError
```

I'm unable to understand why it's not able to generate a series of actions that will take me to the "random navigable point". And if it's not navigable, why is it being returned when I'm trying to find my "random navigable point".

Appreciate any clarification.

yvsriram, 2024-06-21:
Hi Yash,

Are you facing issue with the first episode you try?

If yes, maybe you want to check if goal_pos and current_position are on the same navmesh island? 

For a point, you can use this to get island it is on: https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#get_island-70760

And you can pass island index to `random_navigable_point` to `island_index`

yrraadi-io, 2024-06-22:
Thank you for the pointers. I will try this. Yes I am facing the issue with the first episode I try.

As a follow up, I tried generating a `get_random_navigable_point_near` my agent's current position (`state.position`). I tried this on one episode (790) and then used `find_path` to get a series of actions. That worked, it gave me a list of 30 actions or so. But, upon following those actions I ended up at a different point as the random navigable point near that I generated. I know this since I logged those points.

Happy to share more details if you feel that will be helpful!

Yonggie, 2025-11-10:
Hi, does this means my navmesh is not working? Cus I tried millions and increased the radius but can't find the navigable point...

[image]

[image]