facebookresearch/habitat-sim #2424: Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total

28dvd28, 2024-07-08:
Hello, I'm trying to use habitat-sim for my project, but when i'm running it i'm getting the error:
         Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
I've looked in some other already opened issue, but I can't find a solution, maybe also because i'm working on a WSL and not directly on linux machine. 

The fact that i am using the WSL can it be the reason of why I'm getting the error? If yes can you help me solving this?

aclegg3, 2024-07-10:
Hey @28dvd28 

I assume this can work, but it sounds like you'll need to do some work to expose and setup the GPU correctly.

I saw this: https://canonical-ubuntu-wsl.readthedocs-hosted.com/en/latest/tutorials/gpu-cuda/

Also take a look at our [troubleshooting guide](https://github.com/facebookresearch/habitat-lab/blob/main/TROUBLESHOOTING.md) for graphics issues.

Once you've in a bit and can provide some more info maybe myself or @0mdc can help with any questions.

28dvd28, 2024-07-11:
Hello, 

many thanks for the resources that you gave me.

I've looked at both the link that you have sent to me. From the first one I have a question. I need to download the nvidia driver for windows. But i've already done it for gaming, so i downloaded the _game ready driver_ (I think, it has been i while). Do I need to download the _Studio Driver_?

I've looked to the second link and i've followed the instruction, but i don't know if i have the right output, so:

- running _eglinfo_ I get
[image]
Above is a lot of more output, similar to the last lines and then I get Device platofrom without nothing, is it a problem?
-apt list --installed | grep libglvnd get me 
[image]
is it ok?
-I had to create the file _10_nvidia.json_ in _/usr/share/glvnd/egl_vendor.d/_, the path that i have to put in "library_path" must be like _"../../../../lib/x86_64-linux-gnu/libEGL.so.1"_ or like _"lib/x86_64-linux-gnu/libEGL.so.1"_?

In any case i am still having the issue also testing it in the habitat-lab reposotory with the command _python examples/example.py_. Hope you will be able to help me.

Kind regards

wsychiu, 2024-09-02:
Hello @28dvd28 - did you managed to find a fix for this? I am literally having the same issue as you are.

I can run the provided example `examples/viewer.py` with the `skokloster-castle.glb` example, but cannot get it to work with Matterport3D dataset (after following the [Matterport3D instructions](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#matterport3d-mp3d-dataset)).

`nvidia-smi` and `nvcc --version` are all returning up-to-date drivers and toolkit. I also get the same `eglinfo: eglInitialize failed` after a wall of text. 

Hoping someone has a fix as I can't seem to find many solved problems online...

28dvd28, 2024-09-02:
Hello @wsychiu, I wasn't able to solve this issue, I think that maybe for some strange reason it is needed a full Linux OS on your machine and with just the WSL, for some strange reason, it is not working.

wsychiu, 2024-09-02:
Thanks @28dvd28 - did you get it working on a full Linux install?

28dvd28, 2024-09-02:
@wsychiu - Since it was a school project, I thought that it was too much effort change the OS of my PC, so i did not go on trying to solve it, but the PhD student, that was following me on this project, used a Linux OS and on his PC everything worked fine.

Sebo-the-tramp, 2025-03-01:
I might be late to the party, but I struggled for about a day with this problem and finally figured out a solution.
In my case, I am running the code on a cluster with NO sudo access, and the drivers have already been installed.
After going around looking at issues in particular [issue 1671](https://github.com/facebookresearch/habitat-sim/issues/1671) and [issue 288](https://github.com/facebookresearch/habitat-sim/issues/288) I figured out that the habitat simulator needs a particular file to be placed in a particular place: 

`/usr/share/glvnd/egl_vendor.d/10_nvidia.json` 

turns out I didn't have it...

Therefore I downloaded it from [https://github.com/openai/mujoco-py/blob/master/vendor/10_nvidia.json](https://github.com/openai/mujoco-py/blob/master/vendor/10_nvidia.json) as suggested from [issue 288](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-592584218) to a place that I will not forget and then since I have no sudo access, the only option was to export the os variable as follows:

```bash
export __EGL_VENDOR_LIBRARY_FILENAMES=/the/path/i/will/not/forget/10_nvidia.json
```

After that, everything works for me.

fangbq, 2025-09-25:
I also came across this problem on a remote linux server with 4090 gpu, but without a display.
Tried @Sebo-the-tramp 's solution, does not work for me.

The only way out for me is adding a line of code to use cpu for rendering rather than gpu.

1nibeloflight2, 2025-09-26:
> I might be late to the party, but I struggled for about a day with this problem and finally figured out a solution. In my case, I am running the code on a cluster with NO sudo access, and the drivers have already been installed. After going around looking at issues in particular [issue 1671](https://github.com/facebookresearch/habitat-sim/issues/1671) and [issue 288](https://github.com/facebookresearch/habitat-sim/issues/288) I figured out that the habitat simulator needs a particular file to be placed in a particular place:
> 
> `/usr/share/glvnd/egl_vendor.d/10_nvidia.json`
> 
> turns out I didn't have it...
> 
> Therefore I downloaded it from https://github.com/openai/mujoco-py/blob/master/vendor/10_nvidia.json as suggested from [issue 288](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-592584218) to a place that I will not forget and then since I have no sudo access, the only option was to export the os variable as follows:
> 
> export __EGL_VENDOR_LIBRARY_FILENAMES=/the/path/i/will/not/forget/10_nvidia.json
> 
> After that, everything works for me.

I have /usr/share/glvnd/egl vendor.d/10 nvidia.json, but also be error ,then I download 10_nvidia.json to another dir, 
use 
> export __EGL_VENDOR_LIBRARY_FILENAMES=/the/path/i/will/not/forget/10_nvidia.json
to my dir. and work !!!Perhaps it's because my environment doesn't have the permission to access the file "usr/share/glvnd/egl vendor.d/10 nvidia.json"