facebookresearch/habitat-sim #2428: Gibson dataset glb files

tayyabkhalil-313, 2024-07-12:
## Habitat-Sim version
v0.2.1

## ❓ Questions and Help
I have downloaded the gibson tiny dataset from [here](https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform?pli=1) as gibson_tiny.tar.gz which contained some obj files and others for each scene but no glb files. I then downloaded the [3DSceneGraph](https://redivis.com/datasets/1kf9-cfjvtqc7q/files/x446-7vfbw44r2.xgbruWHmii7SlVbLswFDuA) and then converted it via 
`tools/gen_gibson_semantics.sh /path/to/3DSceneGraph_medium/automated_graph /path/to/GibsonDataset /path/to/output` which produced some scns and ids. 
But I need the glb files for the scenes to run the [demo notebook](https://github.com/rllab-snu/TopologicalSemanticGraphMemory/blob/master/demo/build_tsgm_demo.ipynb?short_path=43d5220#L180 ) and I don't understand how or where to obtain those. Any directions on how to get those glb files?

tayyabkhalil-313, 2024-07-15:
Thank you for responding. The demo notebook is using the gibson_tiny split. Will the gibson_habitat.zip you mentioned also have the same scene files?

aclegg3, 2024-07-15:
I believe the habitat version should contain all the original gibson splits. If that doesn't end up being the case please let me know here.

tayyabkhalil-313, 2024-07-18:
Yes, downloading the habitat version indeed solved the issue. The scene is successfully loaded but when I try to get the objects from semantic scene `scene_objects = sim.semantic_scene.objects` it returns none. The full code is [here](
https://github.com/rllab-snu/TopologicalSemanticGraphMemory/blob/bab0daf402a41b41f3fd2c6680f88b30e81b3dad/demo/build_tsgm_demo.ipynb#L203).