facebookresearch/habitat-sim #2429: Question about own scene rotation in Habitat 3.0

zoeyliu1999, 2024-07-14:
## Habitat-Sim version
v0.3.1

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hi, thank you for your excellent work! I am trying to import my own scene([bedroom_w_light.glb.zip](https://github.com/user-attachments/files/16227823/bedroom_w_light.glb.zip)) into Habitat 3.0, but the scene appears rotated incorrectly (please see the image below). 

[image]

Based on previous issues, I suspect this might be related to the gravity axis. To address this, I tried rotating the entire scene by 90 degrees along the x-axis in Unity, but unfortunately, this did not resolve the issue.

Below is the script I used for adapting my own scene. Could you please provide some advice on how to fix this rotation problem? Thank you!

```python
import gym
import magnum as mn
import numpy as np
import pytest
from omegaconf import DictConfig

import habitat.articulated_agents.humanoids.kinematic_humanoid as kinematic_humanoid
import habitat_sim
import habitat_sim.agent
from habitat.articulated_agent_controllers import (
    HumanoidRearrangeController,
    HumanoidSeqPoseController,
)

default_sim_settings = {
    # settings shared by example.py and benchmark.py
    "max_frames": 1000,
    "width": 640,
    "height": 480,
    "default_agent": 0,
    "sensor_height": 1.5,
    "hfov": 90,
    "color_sensor": True,  # RGB sensor (default: ON)
    "semantic_sensor": False,  # semantic sensor (default: OFF)
    "depth_sensor": False,  # depth sensor (default: OFF)
    "seed": 1,
    "silent": False,  # do not print log info (default: OFF)
    # settings exclusive to example.py
    "save_png": False,  # save the pngs to disk (default: OFF)
    "print_semantic_scene": False,
    "print_semantic_mask_stats": False,
    "compute_shortest_path": False,
    "compute_action_shortest_path": False,
    "scene": "data/scene_datasets/habitat-test-scenes/skokloster-castle.glb",
    "test_scene_data_url": None,
    # "test_scene_data_url": "http://dl.fbaipublicfiles.com/habitat/habitat-test-scenes.zip",
    "goal_position": [5.047, 0.199, 11.145],
    "enable_physics": False,
    "enable_gfx_replay_save": False,
    "physics_config_file": "./data/default.physics_config.json",
    "num_objects": 10,
    "test_object_index": 0,
    "frustum_culling": True,
}

[... 206 lines cut ...]

        # produce some test debug video
        if produce_debug_video:
            from habitat_sim.utils import viz_utils as vut

            motion_name = "test_humanoid_seqpose_controller"
            video_name = (
                os.path.join(output_dir, "scene_video_")
                + motion_name
            )
            vut.make_video(
                observations,
                "color_sensor",
                "color",
                video_name,
                open_vid=False,
            )

if __name__ == "__main__":

    humanoid_name = "female_2"
    scene_path = "bedroom_w_light_x90.glb"
    test_humanoid_seqpose_controller(humanoid_name, scene_path)

```

aclegg3, 2024-07-15:
Hey @zoeyliu1999,

I agree this looks like a simple axis alignment issue.
Habitat is Y-up and -Z forward. When I open your .glb asset in Blender, I see it is aligned with Zup. If I re-export from Blender with the "+Y up" flag **unchecked** I get a glb which will loading into Habitat correctly by default. ([correct_mesh.zip](https://github.com/user-attachments/files/16236949/correct_mesh.zip))

[image]

Alternatively, you could also fix this by using the "up" and "front" vector fields of a [stage config](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stageattributes) if, for example, you could not modify the asset itself for some reason.

closed 2024-07-15 by aclegg3

zoeyliu1999, 2024-07-15:
Hey @aclegg3, thank you for your kind reply!

Regarding the "up" and "front" vector fields of a [stage config](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stageattributes), do I need to construct an entire dataset with all `json` file equipped for this? As you can see, I currently only generate a single scene `.glb` file without any config file.

I kind of prefer not to modify the asset, as importing it into Blender for rotation may cause some furniture textures to go missing, and I am unsure if I can correct this error. However, I am also concerned that constructing a complete dataset with all necessary `json` files might require more effort. Could you please provide some advice? Thanks!

aclegg3, 2024-07-15:
Hey @zoeyliu1999 you don't need to construct a complete dataset. For this use case, you should be able to get away with only the stage_config.json file. Just set the `scene_id` or `settings["scene"]` to point to that config file instead of the .glb.

The minimal contents of a stage config include only the render asset path (relative to the stage config file). Then the "up" and "front" can be adjusted.

This one works to re-orient the scene for me (in the same directory): 
[bedroom_w_light.stage_config.json](https://github.com/user-attachments/files/16237625/bedroom_w_light.stage_config.json)

Contents for reference:
```
{
	"render_asset": "bedroom_w_light.glb",
	"up":[0,1,0],
	"front":[0,0,-1],
	"requires_lighting":true
}

```

aclegg3, 2024-07-15:
No problem, glad it's working for you. Feel free to open new issues if you encounter other challenges and best of luck on your goals.