facebookresearch/habitat-sim #2438: Headless version installed but `DISPLAY not detected ...`

Eureka2001, 2024-08-01:
## System information

Python: 3.7.12
OS: Ubuntu 18.04.4 LTS
Habitat-Lab: [0.2.1-bc85d09](https://github.com/facebookresearch/habitat-lab/tree/bc85d0961cef3b4a08bc9263869606109fb6ff0a)

## ❓ Questions and Help

I install habitat-sim by following [Recommended Conda Installation](https://github.com/facebookresearch/habitat-sim?tab=readme-ov-file#recommended-conda-packages).

```
conda install habitat-sim=0.2.1 headless -c conda-forge -c aihabitat
```

I am sure that I used the right instruction since I have checked the [related habitat-lab issue](https://github.com/facebookresearch/habitat-lab/issues/1888) and saw the information from conda as follows:

```
## Package Plan ##

  environment location: /home/XXX/miniconda3/envs/XXX

  added / updated specs:
    - habitat-sim=0.2.1
    - headless
```

But I meet error: `DISPLAY not detected. For headless systems, compile with --headless for EGL support`.

aclegg3, 2024-08-12:
Where are you getting that error? It is after installation or during?

Eureka2001, 2024-08-13:
After the installation. I encountered the error when I ran code.

I have resolved this issue by downloading the `linux-64/habitat-sim-0.2.1-py3.7_headless` Anaconda package from [here](https://anaconda.org/aihabitat/habitat-sim/files?page=2) and installing it locally.

aclegg3, 2024-08-13:
I'm glad you figured this out. Perhaps you had a stale installation on the environment? In any case, I'll close the issue now, glad to hear it is working for you!

closed 2024-08-13 by aclegg3

licongfei001, 2025-06-27:
@Eureka2001 Hi, I am a newer of conda. How do you install linux-64/habitat-sim-0.2.1-py3.7_headless?