facebookresearch/habitat-sim #2440: Can the agent move along the Y-axis?

hjy5311, 2024-08-06:
To make the agent function like a drone, I need it to move along the Y-axis (UP-DOWN direction). I attempted to achieve this using the MoveUp control function registered in [default_controls.py](https://github.com/facebookresearch/habitat-sim/blob/main/src_python/habitat_sim/agent/controls/default_controls.py), originally designed to raise a sensor with `body_action `set to `False`. Now, I've set `body_action=True` to move the entire agent (including the sensor) along the Y-axis, but the execution doesn't meet expectations—the agent's position remains unchanged. I'm wondering if there are specific mechanisms restricting the agent's movement along the Y-axis and if there are methods to enable movement in this direction.

`@registry.register_move_fn(body_action=True)
class MoveUp(SceneNodeControl):
    def __call__(self, scene_node: SceneNode, actuation_spec: ActuationSpec) -> None:
        _move_along(scene_node, actuation_spec.amount, _Y_AXIS)`

aclegg3, 2024-08-06:
Hey @hjy5311 
When using the default movement actions provided with the pointnav task, the Agent's SceneNode (3D transform) is locked to the navmesh. There is a "step_filter" which snaps the Agent to the navmesh after any translation.

You could write custom logic and use a custom action set which does not use the step filter, but you will need to handle the collision detection yourself since the navmesh abstraction doesn't really extend well to flying robots (as they are not restricted to navigable floor regions).

My suggestion is to implement your action set using the collision detection functionality, providing a 3D mesh to represent the robot's body and checking that this body is not intersecting with the environment after each requested discrete motion.

Check out this tutorial (https://github.com/facebookresearch/habitat-sim/blob/main/examples/tutorials/notebooks/ECCV_2020_Interactivity.ipynb) for an example of loading a body, collision detection, velocity control and other useful mechanisms.

aclegg3, 2024-08-06:
Also, I did write a quick and dirty example of a quad-rotor several years ago. Likely the API has shifted significantly, but maybe the reference is still useful: https://github.com/facebookresearch/habitat-sim/commit/c4a4c53345e041677c17b1f502d150666465974e

hjy5311, 2024-08-11:
Thank you very much for your answer. It has been very helpful. I am now considering using an agent to detect if it intersects with the environment. I would like to know if Habitat-sim provides a direct interface at the simulator level for obtaining the bounding boxes or other attributes that reflect the volume size of objects and the scene. This will impact the difficulty of adding collision detection. 

By reading the code, I found that there are interfaces available for obtaining the volume information of objects. However, for the scene, I have not yet discovered  code that provides access to the volume attributes related to walls, scene boundaries, etc.

Thank you for your answer.

aclegg3, 2024-08-12:
Hey @hjy5311 

You should check out the rigid object tutorial from our[ tutorial series](https://aihabitat.org/docs/resource_index/#tutorials). It covers topics related to use of physics and collision detection.

To quickly answer your question: yes, Habitat provides bounding box and collision detection routines which you can use to determine if your agent object is in collision with the scene or other objects.

AABB: https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedRigidObject.html#aabb
Object centric binary contact test: https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedRigidObject.html#contact_test
Full scene collision detection: https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#perform_discrete_collision_detection
Contact point data available after collision detection: https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#get_physics_contact_points