facebookresearch/habitat-sim #2441: How to set a directional sound source?

szhangjc, 2024-08-07:
## ❓ Questions and Help
Thanks for your excellent work! I set my sound source as shown below,
`source=np.array([0.0,4.0,1.7]) 
audio_sensor = sim.get_agent(0)._sensors["audio_sensor"]
audio_sensor.setAudioSourceTransform(source) 
audio_sensor.setAudioMaterialsJSON("data/mp3d_material_config.json") 
`
I wonder whether the sound source is omnidirectional or directional? If it is directional, how to change the direction of the sound source?Thanks for your answer!

aclegg3, 2024-08-12:
Hey @szhangjc,

I believe all sources are current omni-directional.
Please check out the docs here: https://github.com/facebookresearch/habitat-sim/blob/main/docs/AUDIO.md

szhangjc, 2024-08-13:
I got it, thanks for your answer.

closed 2024-08-13 by szhangjc