facebookresearch/habitat-sim #2443: [None Docker and headless install in conda enviroment] unable to find EGL device for CUDA device 0

TLvCo, 2024-08-08:
## Habitat-Sim version
v0.1.7

conda install -c aihabitat -c conda-forge habitat-sim=0.1.7 headless
git clone --branch v0.1.7 git@github.com:facebookresearch/habitat-lab.git
cd habitat-lab
python setup.py develop --all # install habitat and habitat_baselines

I had installed openGL, and successed to use EGL to show UI in another conda enviroment(python3.8 torch2.1.0+cu121)
but errors occupied when I used habitat-smi in a new **conda** eviroment(python3.6 torch1.9.1+cu111)

## ❓ Questions and Help

```shell
I0808 14:58:27.797049 2805534 Simulator.cpp:156] Loading navmesh from data/andong/Matterport/v1/mp3d/17DRP5sb8fy/17DRP5sb8fy.navmesh
I0808 14:58:27.797112 2805534 Simulator.cpp:158] Loaded.
I0808 14:58:27.797122 2805534 SceneGraph.h:93] Created DrawableGroup:
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_EXT_device_drm EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=1): EGL_EXT_device_drm EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=2): EGL_EXT_device_drm EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=3): EGL_MESA_device_software EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context
```

aclegg3, 2024-08-12:
@0mdc thoughts?

nicklhy, 2024-08-15:
Same problem here.

bytepioneerX, 2025-03-06:
If you use  habitat-sim in the docker or WSL:

1. Install somthing essential

apt-get install -y  libx11-dev  libgles2-mesa  libgles2-mesa-dev  gnupg2  libbullet-dev && rm -rf /var/lib/apt/lists/*

2. Down habitat-sim headlss version from [this](https://anaconda.org/aihabitat/habitat-sim/files)

wegt https://anaconda.org/aihabitat/habitat-sim/0.1.7/download/linux-64/habitat-sim-0.1.7-py3.7_headless_bullet_linux_856d4b08c1a2632626bf0d205bf46471a99502b7.tar.bz2  && 
conda install /tmp/habitat-sim-0.1.7-py3.7_headless_bullet_linux_856d4b08c1a2632626bf0d205bf46471a99502b7.tar.bz2 

4. echo "unset DISPLAY" >> ~/.bashrc && source ~/.bashrc

5. enjoy!

FengMengguang, 2025-04-18:
anyone know how to fix the problem

bytepioneerX, 2025-04-18:
This is an automatic reply email from 闫倬豪.I have received your email, thank you!