facebookresearch/habitat-sim #2454: No Glob path result found for *.json so unable to load templates from that path.

anilkusc, 2024-08-20:
## Habitat-Sim version: v0.3.1
I've installed habitat-sim and habitat lab. I can run examples without issue.
I want to use hm3d dataset but I am have error when I try.
I am following the [tutorial](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#habitat-matterport-3d-research-dataset-hm3d) 
I can download the dataset successfully: 
`python -m habitat_sim.utils.datasets_download --username <api-token-id> --password <api-token-secret> --uids hm3d_minival_v0.2`
I am creating a test python file in examples folder to see scenes:

>#examples/test.py 
import habitat_sim
backend_cfg = habitat_sim.SimulatorConfiguration()
backend_cfg.scene_id = "<PATH TO HM3D>/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb"
backend_cfg.scene_dataset_config_file = "<PATH TO HM3D>/hm3d_annotated_basis.scene_dataset_config.json"
sem_cfg = habitat_sim.CameraSensorSpec()
sem_cfg.uuid = "semantic"
sem_cfg.sensor_type = habitat_sim.SensorType.SEMANTIC
agent_cfg = habitat_sim.agent.AgentConfiguration()
agent_cfg.sensor_specifications = [sem_cfg]
sim_cfg = habitat_sim.Configuration(backend_cfg, [agent_cfg])
sim = habitat_sim.Simulator(sim_cfg)

When I try to run I got following error:
>.
.
[13:07:07:929709]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/habitat-sim/data/scene_datasets/hm3d/val/00891-cvZr5TUy5C5/*.basis.scene_instance.json` so unable to load templates from that path.
[13:07:07:929722]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/habitat-sim/data/scene_datasets/hm3d/val/00894-HY1NcmCgn3n/*.basis.scene_instance.json` so unable to load templates from that path.
[13:07:10:986678]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset : Dataset : 'hm3d_annotated_basis' : Lighting Layout Attributes 'no_lights' specified in Scene Attributes but does not exist in dataset, so creating default.

When I check dataset folder I can not find any *basis.scene_instance.json file.
In the tutorial it says:

>Most importantly, ensure that the hm3d_annotated_*, *.semantic.glb, and *.semantic.txt files are present.

When I check all mentioned files represent.

So do you have a further documentation or tutorial to use hm3d with habitat on hand ?

albertcity, 2024-08-22:
> ## Habitat-Sim version: v0.3.1
> I've installed habitat-sim and habitat lab. I can run examples without issue. I want to use hm3d dataset but I am have error when I try. I am following the [tutorial](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#habitat-matterport-3d-research-dataset-hm3d) I can download the dataset successfully: `python -m habitat_sim.utils.datasets_download --username <api-token-id> --password <api-token-secret> --uids hm3d_minival_v0.2` I am creating a test python file in examples folder to see scenes:
> 
> > #examples/test.py
> > import habitat_sim
> > backend_cfg = habitat_sim.SimulatorConfiguration()
> > backend_cfg.scene_id = "/habitat-sim/data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb"
> > backend_cfg.scene_dataset_config_file = "/habitat-sim/data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json"
> > sem_cfg = habitat_sim.CameraSensorSpec()
> > sem_cfg.uuid = "semantic"
> > sem_cfg.sensor_type = habitat_sim.SensorType.SEMANTIC
> > agent_cfg = habitat_sim.agent.AgentConfiguration()
> > agent_cfg.sensor_specifications = [sem_cfg]
> > sim_cfg = habitat_sim.Configuration(backend_cfg, [agent_cfg])
> > sim = habitat_sim.Simulator(sim_cfg)
> 
> When I try to run I got following error:
> 
> > .
> > .
> > [13:07:07:929709]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad :  : No Glob path result found for `/habitat-sim/data/scene_datasets/hm3d/val/00891-cvZr5TUy5C5/*.basis.scene_instance.json` so unable to load templates from that path.
> > [13:07:07:929722]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad :  : No Glob path result found for `/habitat-sim/data/scene_datasets/hm3d/val/00894-HY1NcmCgn3n/*.basis.scene_instance.json` so unable to load templates from that path.
> > [13:07:10:986678]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset : Dataset : 'hm3d_annotated_basis' : Lighting Layout Attributes 'no_lights' specified in Scene Attributes but does not exist in dataset, so creating default.
> 
> When I check dataset folder I can not find any *basis.scene_instance.json file. In the tutorial it says:
> 
> > Most importantly, ensure that the hm3d_annotated_*, *.semantic.glb, and *.semantic.txt files are present.
> 
> When I check all mentioned files represent.
> 
> So do you have a further documentation or tutorial to use hm3d with habitat on hand ?

I have met the same problem, any solutions?

hjy0712, 2025-01-13:
i have the same problem, too
`[19:29:32:458830]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `./data/scene_datasets/hm3d_v0.2/val/00894-HY1NcmCgn3n/*.basis.scene_instance.json` so unable to load templates from that path.`

Roberyan, 2025-02-07:
I came across the same issue, does anyone know a solution? I am using the demo code, with all data downloaded
```
import habitat_sim

PATH_TO_HM3D = "./data/scene_datasets/hm3d"

backend_cfg = habitat_sim.SimulatorConfiguration()
backend_cfg.scene_id = f"{PATH_TO_HM3D}/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb"
backend_cfg.scene_dataset_config_file = f"{PATH_TO_HM3D}/hm3d_annotated_basis.scene_dataset_config.json"

sem_cfg = habitat_sim.CameraSensorSpec()
sem_cfg.uuid = "semantic"
sem_cfg.sensor_type = habitat_sim.SensorType.SEMANTIC

agent_cfg = habitat_sim.agent.AgentConfiguration()
agent_cfg.sensor_specifications = [sem_cfg]

sim_cfg = habitat_sim.Configuration(backend_cfg, [agent_cfg])
sim = habitat_sim.Simulator(sim_cfg)
```

and I got error:
```
(habitat) marmot@Temari:~/Boyang/pvlm$ python pretrain_data_collection/load_hm3d.py 
[23:36:52:598459]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/example/00861-GLAQ4DNUx5U/*.basis.glb` so unable to load templates from that path.
[23:36:52:598735]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00900-XRJ2muKkwAV/*.basis.glb` so unable to load templates from that path.
[23:36:52:598749]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00901-aJoC8Qw6xQ5/*.basis.glb` so unable to load templates from that path.
[23:36:52:598759]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00904-S9CUp5RsFY9/*.basis.glb` so unable to load templates from that path.
[23:36:52:598767]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00905-9v5HzFAB1cm/*.basis.glb` so unable to load templates from that path.
[23:36:52:598776]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00907-WGxjrSp71zX/*.basis.glb` so unable to load templates from that path.
[23:36:52:598785]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00909-MrKys1cv4jD/*.basis.glb` so unable to load templates from that path.
[23:36:52:598793]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00910-jE7Qzu3FyJD/*.basis.glb` so unable to load templates from that path.
[23:36:52:598801]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00911-a8h4mZT8buN/*.basis.glb` so unable to load templates from that path.
[23:36:52:598810]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00913-URu5ybF8PrH/*.basis.glb` so unable to load templates from that path.
[23:36:52:598818]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00914-TxyEAAVD4iD/*.basis.glb` so unable to load templates from that path.
[23:36:52:598827]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00915-ieavb8rbmA9/*.basis.glb` so unable to load templates from that path.
[23:36:52:598836]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00918-NQX2SZD1TMJ/*.basis.glb` so unable to load templates from that path.
[23:36:52:598844]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00921-nhtTwUrWv7v/*.basis.glb` so unable to load templates from that path.
[23:36:52:598853]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00925-4QUpJVDQ1DN/*.basis.glb` so unable to load templates from that path.
[23:36:52:598861]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00926-egVUByjgpHU/*.basis.glb` so unable to load templates from that path.
[23:36:52:598871]:[Warning]:[Metadata] AbstractAttributesManager.h(524)::buildAttrSrcPathsFromJSONAndLoad : <Stage Template> : No Glob path result found for `./data/scene_datasets/hm3d/test/00927-kucQY8TbKkz/*.basis.glb` so unable to load templates from that path.
[... cut ...]
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
[23:36:52:660762]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset : Dataset : 'hm3d_annotated_basis' : Lighting Layout Attributes 'no_lights' specified in Scene Attributes but does not exist in dataset, so creating default.
```

LiNY2-01, 2025-02-15:
i have the same problem, too

```
[14:25:14:916381]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00904-S9CUp5RsFY9/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916389]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00905-9v5HzFAB1cm/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916398]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00907-WGxjrSp71zX/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916406]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00909-MrKys1cv4jD/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916414]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00910-jE7Qzu3FyJD/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916422]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00911-a8h4mZT8buN/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916431]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00913-URu5ybF8PrH/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916439]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00914-TxyEAAVD4iD/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916447]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00915-ieavb8rbmA9/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916455]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00918-NQX2SZD1TMJ/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916463]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00921-nhtTwUrWv7v/*.basis.scene_instance.json` so unable to load templates from that path.
[14:25:14:916471]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `/data/hm3d/tars/data/scene_datasets/hm3d/test/00925-4QUpJVDQ1DN/*.basis.scene_instance.json` so unable to load templates from that path.
```

yunho-c, 2025-02-16:
I have had the same error for a few months. One hacky workaround it (probably only possible in Jupyter environments) is to interrupt the execution and start again—the error does not occur for subsequent runs. Quick demonstration at https://www.youtube.com/watch?v=U3qbSrslrDs.

Ritabrata-Chakraborty, 2025-07-15:
@anilkusc I am facing the exact same issue. In my case, the basis.scene_instance.json files are not even downloaded. How do I download the full dataset for val_0.2