facebookresearch/habitat-sim #2480: download hm3d_basis.scene_dataset_config.json

kunhumllop, 2024-10-15:
Hello, I cannot download the file (hm3d_basis.scene_dataset_config.json). Can you provide a download link? Thank you very much!

Eku127, 2024-10-15:
Hi, do you have the link for downloading this file? I could not find this file in https://github.com/matterport/habitat-matterport-3dresearch?tab=readme-ov-file

kunhumllop, 2024-10-15:
Yeah, I can't download the file either.

aclegg3, 2024-10-16:
Hey folks, that file is found in `hm3d-minival-semantic-configs-v0.2.tar` (Format == semantic-configs).

closed 2024-10-16 by aclegg3

Lazy-boy193, 2024-11-05:
> 嘿伙计们，该文件位于`hm3d-minival-semantic-configs-v0.2.tar`（格式 == semantic-configs）。

`> ls <PATH TO HM3D>
hm3d_annotated_basis.scene_dataset_config.json    train/
hm3d_basis.scene_dataset_config.json              val/
minival/
`

What's the meaning of this?
Does it mean that the `hm3d_annotated-basis.scene_dataset_comfig.rson` file is actually the `hm3d_masis.scene_dataset_comfig.rson` file? So does the above structure only need to satisfy the following structure:

`> ls <PATH TO HM3D>
hm3d_annotated_basis.scene_dataset_config.json   
train/
val/
minival/
`

Excuse me, is that what the above answer means?